hn/ginlong-solis #3: solis-rhi-3k-48es-5g support

chrgraf, 2023-04-30:
Dear Hajo,

your code is genius. Thanks for this. Surely beyond my own skills..
Planning to fork your code and add MQTT - at least MQTT is running already in mine testbed successful

I realized my Solis Inverter (solis-rhi-3k-48es-5g) does not get detected, hence the script did not work as expected:
Register 35000 delivers: 2031--- 1 phase LV AC Couple energy storage inverter

Made those changes to accomodate:

    case SDT_ITYPE:  /* Solis inverter type definition, switches lookup table */
        {
          char buf[2 + 4 + 1];
          unsigned int regvalue = modbus.getResponseBuffer(0);
          
          sprintf(buf, "0x%04X", regvalue);
          Serial.print(buf);

          if ((regvalue / 100) == 10) {           /* RS485_MODBUS (INV-3000ID EPM-36000ID) inverter protocol */
            solis = solisINV;
          } else if ((regvalue / 100) == 20) {    /* RS485_MODBUS (ESINV-33000ID) energy storage inverter protocol */
            solis = solisESINV;
          } else if ((regvalue) == 8241) {    /* RS485_MODBUS solis-rhi-3k-48es-5g */
            solis = solisESINV;
             
          }
        }
        break;

Reading of Serial somehow fails as well. Still looking into this and will you keep updated.

Again, thanks for this execellent piece of work

thanks

hn, 2023-05-01:
I appreciate that you like the code.

So register 35000 of your inverter is hex 0x20XX, that's interesting.

If you use Home Assistant, interfacing the inverter with ESPhome would be an option without having to fiddle with source code.

chrgraf, 2023-05-17:
Kind thanks Hajo – indeed it makes very much sense to become familiar with ESPhome. Doing my first baby-steps..

My Solis-Inverter is heavily overshooting when some excess-power is available from the PV. It charges the battery with to much power and takes lots from grid due to over-excessive charging. Had seen that the Solis struggelt for minutes to get his control-loop adjusted.  Just trying to see if modifying ramp-rate could solve it.

But I am quite sure that setting register 43117 would solve it as well to prohibit the solis charging to heavy.

3
43117
U16
0.1A
Battery charge current maximum setting
float

(taken from https://github.com/peufeu2/GrugBus)

Point is, that I am lacking the skill to craft the correct modbus rtu frame. Wondering if you could help-out with some code to get exactly this register written to an arbitrary value? I shall be able with my skills to get then the remaning built around..

Thanks

[External Email. Be cautious of content]

Juniper Business Use Only

hn, 2023-05-17:
With ESPhome, this should simply be possible with [Controller Number](https://esphome.io/components/number/modbus_controller.html). You will probably have to use an FP32 number type.

If you're more into handcrafting, you may want to look at [the RTC update](https://github.com/hn/ginlong-solis/blob/master/solis-modbus-inv.yaml#L338) lambda.

chrgraf, 2023-05-17:
Many Thanks Hajo.

Will give it a shot. Apppreciate your help

Thanks

morrisonj, 2023-05-17:
> your code is genius. Thanks for this. Surely beyond my own skills.. Planning to fork your code and add MQTT - at least MQTT is running already in mine testbed successful

I'm planning on adding MQTT too. Would you mind sharing your implementation?

chrgraf, 2023-05-20:
Dear Morrisonj,

just started to publish. Still in progress of improving code, cleanup,documenting,..

But here you are:

https://github.com/chrgraf/solis_esp_rs485_mqt

thanks

hn, 2023-05-21:
@chrgraf , great work, thanks for contributing!

If you re-use a significant amount of code, a more Github-like way would be to fork this repository and remove/change things as needed. This makes it easier to forward/backward patch code, fix bugs etc. And: the link in your comment misses a 't' at the end.

chrgraf, 2023-05-21:
Dear Hajo,
I have forked it today.

Will need to shut down the old repo then.
As git is not my strength, let me know what would be best.

Thanks

Senior Consulting SE

***@***.******@***.***>

Mobile: 0175/2929157

www.juniper.net<http://www.juniper.net/>

Juniper Networks GmbH

Sitz der Gesellschaft / Registered office: Oskar-Schlemmer-Str. 15, 80807 München

Registergericht / Commercial registry: München, HRB 205571

Geschäftsführer / Directors: Johannes Albers, Stephen Byrne

hn, 2023-05-22:
@chrgraf There is no right or wrong. If you want to restore your original repository, you can delete files that are unnecessary for your use case or even rename the repository.

And: You might want to clean up your email footer from the Github comments via the web interface, currently one can see your mobile number, among other things.