hn/ginlong-solis #6: Inverter type (register 35000) and model type database

hn, 2023-05-21:
To be able to determine the inverter type automatically, the exact content of ModBus register 35000 is important. There seem to be different interpretations of the documentation, so it would be interesting to collect as many register values as possible here to create a robust detection logic.

If you want to contribute, just post here:

```
Inverter type as printed on the unit:
Type definition (reg 35000):
Model type (reg 3000 for INV, reg 33000 for ESINV):
```

Please pay attention to whether the value is hexadecimal or decimal, this seems to be an inconsistency.

hn, 2023-05-21:
@morrisonj contributed in #5  :

```
Inverter type as printed on the unit: S5-GR3P15K
Type definition (reg 35000): 0x0000
Model type (reg 3000 for INV, reg 33000 for ESINV): 0x0509
```

hn, 2023-05-21:
@chrgraf contributed in #3 :

```
Inverter type as printed on the unit: solis-rhi-3k-48es-5g
Type definition (reg 35000): 0x2031
Model type (reg 3000 for INV, reg 33000 for ESINV): - unknown -
```

richardloxley, 2023-06-19:
There definitely seems to be a discrepancy between hex and decimal, depending on the model!

I am working with two models of inverter at the moment, and this is what I get:

S6-GR1P2.5K-M
Type definition (reg 35000): 1010 decimal = 0x3F2
Model type (reg 3000 for INV): 260 decimal = 0x104
Expected model definition according to documentation: 1010

S5-EH1P6K-L
Type definition (reg 35000): 8240 decimal = 0x2030
Model type (reg 33000 for ESINV): not yet obtained
Expected model definition according to documentation: 2030

The documentation for register 35000 implies that the values should be interpreted as hex (they refer to low 8 bits, high 8 bits, and 0000H).

But it looks like my non-hybrid inverter returns the value as if it was a decimal.  I suspect that's a bug when they wrote the firmware (maybe since corrected?)

As a working proposition, I'm going to take anything <0x2000 as INV and anything >=0x2000 as ESINV.  That's because any hex values written as decimal will be lower than their correct hex values, so provided that none of the hybrid inverters have the same bug, it should work.

TheGroundZero, 2024-04-22:
```
Inverter type as printed on the unit: RHI-3P10K-HVES-5G
Type definition (reg 35000): 2060 (reported by Home Assistant)
Model type (reg 3000 for INV, reg 33000 for ESINV): 00ca (reported by Home Assistant)
```

helmuteke, 2024-05-29:
> There definitely seems to be a discrepancy between hex and decimal, depending on the model!
>
> I am working with two models of inverter at the moment, and this is what I get:
>
> S6-GR1P2.5K-M Type [...]

Hello ,

How did you do it for the S5-EH1P6K-L ?? i have a S5-EH1P5K-L but can not achieve to work

philpem, 2025-04-29:
With ESINV.

Solis S5-EH1P3.6K-L

Inverter type definition: 2030 
Product model: 3102
Inverter SN: 0363000000000000 (seems to be masked like https://github.com/hn/ginlong-solis/issues/1#issuecomment-2212991547 reports)

gigaz, 2025-05-18:
Inverter type as printed on the unit: S5-EH1P5K-L
inverter Type definition : 2030
product Model :  3104
these two values were taken from esphome

closed 2026-11-01 by hn