hn/ginlong-solis #7: Can't upload to S3

hcouplet, 2023-05-22:
I have an error when uploading the firmware :  ValueError: Failed to write to 0x800B000

Here is the log : 

➜  libretiny-esphome git:(platform/libretuya) ✗ python3.10 -m esphome upload solis2-stick.yaml --device /dev/tty.SLAB_USBtoUART
INFO Reading configuration solis2-stick.yaml...
****************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************
Obsolete PIO Core v6.1.6 is used (previous was 6.1.7)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/en/latest/core/installation/troubleshooting.html
****************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************
Processing solis2-stick (board: generic-rtl8710bx-4mb-980k; framework: arduino; platform: libretiny)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Configuring upload protocol...
AVAILABLE: uart
CURRENT: upload_protocol = uart
Looking for upload port...
Using manually specified: /dev/tty.SLAB_USBtoUART
Uploading .pioenvs/solis2-stick/firmware.uf2
|-- Detected file type: UF2 - esphome 2023.5.0-dev
|-- Connecting to 'Realtek AmebaZ' on /dev/tty.SLAB_USBtoUART @ 1500000
|   |-- Success! Chip info: Realtek RTL87xxB
|-- Writing '.pioenvs/solis2-stick/firmware.uf2'
|   |-- esphome 2023.5.0-dev @ 2023-05-23 00:27:42 -> generic-rtl8710bx-4mb-980k
|-- ValueError: Failed to write to 0x800B000
|   |-- File "/usr/local/lib/python3.10/site-packages/ltchiptool/soc/ambz/flash.py", line 147, in flash_write_raw

 My chip looks the same as yours : 
[image]

hn, 2023-05-23:
Please try to solely flash the app alone using the Python2 version of [rtltool](https://esp8266.ru/forum/threads/rtl871xbx-tools-ameba-z.2673/):

```
python2 ./rtltool2.py -p /dev/ttyUSB0 wf 0xb000 ./.esphome/build/solis-inv/.pioenvs/solis-inv/image_0x00B000.ota1.bin
```

hcouplet, 2023-05-24:
Thanks for your response.  It helped a lot.
Now I can flash it and it's working.

Here are my findings.
1. As the link to the russian site for rtltool.py was dead, so I used rtltool.py from https://hackaday.io/project/182205-py-ft10/log/206206-rtltool or https://github.com/cyber-murmel/rtltool.  They are both the same and may be you could update the reference.
2. for the rtltool.py, no need for the "RTL8762x MP Tool Kits(ZIP)".  Just run trltool.py with pyton2.7
3  
I have issue with OTA. It says that the Auth is not correct. 
I had to flash in twice (not sure if the is correct) at  0xb000 and at 0x100000

```
/Users/hcouplet/.pyenv/versions/2.7.18/bin/python2 /Users/hcouplet/prg/projects/other/homeassistant/solis-stick/rtltool.py \
    -p /dev/tty.usbserial-0001 wf 0xb000 \
    /Users/hcouplet/prg/projects/other/homeassistant/solis-stick/libretiny-esphome/.esphome/build/solis2-stick/.pioenvs/solis2-stick/image_ota1.0x00B000.bin

/Users/hcouplet/.pyenv/versions/2.7.18/bin/python2 /Users/hcouplet/prg/projects/other/homeassistant/solis-stick/rtltool.py \
    -p /dev/tty.usbserial-0001 wf 0x100000 \
    /Users/hcouplet/prg/projects/other/homeassistant/solis-stick/libretiny-esphome/.esphome/build/solis2-stick/.pioenvs/solis2-stick/image_ota2.0x100000.bin
````

I ended up disabling OTA Password and OTA is working fine (I needed this before putting back the S3 card into the case to be able to update remotely the firmware).
I updated ota_component.cpp to bypass the wrong ota password commenting line at 239 :
[image]
 Flash is working find but the response is not valid. it's some kind of digest auth with nonce and cnonce.

4. I did all the initial flash on my mac but after installed LibreTiny ESPHOME on my HA. ESPHome was already there and I can use both of them :  
[image]

For this I used the add repository button on https://github.com/libretiny-eu/esphome-hass-addon
So now I am using the libretiny version only for my solis S3.
OTA (without password) is working fine.

hn, 2023-05-24:
- You are Solis S3 stick open source registered user # 3 :)
- Updated [rtltool2 link](https://github.com/pvvx/RTL0B_SDK/blob/master/mkb/rtltool.py). The one I use was created by @pvvx, the ones you mention differ (but probably work just as well)
- I never had any OTA password issues, that's strange.
- Don't know why ltchiptool fails for you, maybe @kuba2k2 has an approach on how to fix/debug this.

closed 2023-06-23 by hn

hn, 2023-06-23:
@kuba2k2

I use this [rtltool](https://github.com/pvvx/RTL0B_SDK/blob/master/mkb/rtltool.py) version for python2, created by @pvvx, It always worked so far.

I re-installed both LibreTiny and ESPhome when you released LT 1.0.2 (May 11). Unlike @hcouplet, I don't have any (OTA-)problems, so everything is fine (except the [strange ModBus sync problem](https://github.com/hn/ginlong-solis/issues/4), but that's a different story).