hn/ginlong-solis #8: Can you update your Solis inverter through this integration?

matthewbyrne, 2023-05-23:
I've heard of older wifi data logger modules being capable of updating settings on their solis inverter remotely (via Home assistant automation for example). Does this integration provide that capability?

hn, 2023-05-23:
In my opinion, there is nothing to stop all the things that can be set via ModBus (holding registers, see INV/ESINV register maps) from also being realised via this solution. For example, [the inverter's internal clock is synchronized via NTP](https://github.com/hn/ginlong-solis/blob/master/solis-modbus-inv.yaml#L329) for INV type inverters.

hcouplet, 2023-05-24:
I have 3 solis inverter and I must limit the power injected to grid to max 10Kw.  (Belgium regulation).

I have a P1 house counter that report the power injected to the grid.
So I am updating inverters power limit to do that.   ( I know that I can limit the power by connecting the meter to the inverter but this way I can define the balance myself)

Here is what is working for a solis S6.
By default it is set to 110% .   One I change to value, it change the power generated.

```
number:
    ## Set ≈ limit
  - platform: modbus_controller
    modbus_controller_id: modbus_master
    id: solis2_set_power_limit
    name: "Set Solis2 Power limit"
    address: 3051
    unit_of_measurement: "%"
    register_type: holding
    value_type: S_WORD
    mode: box
    step: 0.5
    entity_category: config
    skip_updates: 100
    min_value: 0
    max_value: 110
    lambda: "return  x / 100; "
    write_lambda: |-
      int16_t limit = x * 100;
      ESP_LOGI("main", "Set Power Limit %d", limit);

      //Send power limit        
      esphome::modbus_controller::ModbusCommandItem set_payload_commandValue = esphome::modbus_controller::ModbusCommandItem::create_write_single_command(modbus_master, 3051, limit);
      // Submit the command to the send queue
      modbus_master->queue_command(set_payload_commandValue);
      return {};       

```

[image]

hcouplet, 2023-05-24:
Yep it's working fine.

I tried other registries (i.e. 3081 with 3070) to set the absolute limit but they didn't worked.
This one was in the inverter menu and is limited on the inverter to max 110%.  I was also limited to 110% max with the modbus.
I didn't had the 3041, I guess it's by default activated.
Anyway, if the inverter deliver 4000W, setting a value of 10%, instantanely changed the delivery to 500W (I have a 5K inverter).   Also changing on the inverter, changed the value on HA and vice-versa.

I can check tomorrow during day time my 3041 value.
I have still lo check with my solis hybrid : I have two S6-GR1P5K  and one RHI-6K-48ES-5G.  (this is a DIY install).
Once I have the working value, I will post it here.

hcouplet, 2023-05-25:
3041 is set to "No response mode" on my inverter. So the register 3052 don't seem to use the 3041.
Here is the select I did : 
```
select:
  - platform: modbus_controller
    name: "Working Mode"
    address: 3040
    value_type: S_WORD
    optionsmap:
      "No response mode": 00
      "Volt–watt default": 01
      "Volt–var": 02
      "Fixed power factor": 03
      "Fix reactive power": 04
      "Power-PF": 05
      "Rule21Volt–watt": 06
```

For the register 3052, On the inverter it's on the menu : 
Advanced Settings / 0010 / Power Control / Set Output Power 

Maybe the doc is not correct, I am wondering if I am not changing the register describe at 3050 : 
[image]
This would make sense with the multiply by 100 (value max 11000) and it does not depend on other register.

May be we need to do plus +1 instead of -1 as explained in the doc ? : 
[image]

Anyway I confirm that it's working great and the power is adjusted down and up directly : I can go down to 500W and the second later it goes back to 4500W

hn, 2023-05-25:
According to [this Solis FAQ](https://solis-service.solisinverters.com/de/support/solutions/articles/44002032572-control-maximum-active-power-generation) 'Set Output Power' is a volatile Setting (= reset during reboot) and 'output_p mit restore' is a permanent setting. Maybe each of this settings is set in different ModBus registers.

hcouplet, 2023-05-25:
This is good to know. The one I am changing is the 'Set Output Power' and not the one with restore as also available in the inverter menu.  For what I want the one in RAM is better so I can update it often without risking to burn the flash with too many write.

hn, 2023-05-25:
@hcouplet Just pushed https://github.com/hn/ginlong-solis/commit/44354154ec5fdc084cd8949501f590db02b61f17, which includes your code (with small modifications, I think the lambdas aren't needed).

hcouplet, 2023-05-30:
Here is the code for solis hybrid. I tested on Solis RHI-6K-48ES-5G (firmware 3D0037).

To change output power. This is the value in RAM, so there is no write count limitation and it will be gone after reboot.

number:
  - platform: modbus_controller
    modbus_controller_id: modbus_master
    id: set_power_limit_fixed
    name: "Set Power limit"
    register_type: holding
    address: 43052
    value_type: U_WORD
    unit_of_measurement: '%'
    entity_category: config
    icon: mdi:percent
    skip_updates: 10
    mode: box
    min_value: 0
    max_value: 110
    multiply: 100

Grid AC voltage is also useful for me : 
  - platform: modbus_controller
    modbus_controller_id: modbus_master
    name: AC Voltage
    device_class: voltage
    state_class: measurement
    unit_of_measurement: V
    register_type: read
    address: 33073
    value_type: U_WORD
    accuracy_decimals: 1
    filters:
      - multiply: 0.1   

For S6 non hybrid, I am using : 

  - platform: modbus_controller
    modbus_controller_id: modbus_master
    name: AC Voltage
    device_class: voltage
    state_class: measurement
    unit_of_measurement: V
    register_type: read
    address: 3035 # = 3036 - 1
    value_type: U_WORD
    accuracy_decimals: 1
    filters:
      - multiply: 0.1

Here is the doc I used for the hybrid : 
[RS485_MODBUS-Hybrid-BACoghlan-201811228-1854.pdf](https://github.com/hn/ginlong-solis/files/11604329/RS485_MODBUS-Hybrid-BACoghlan-201811228-1854.pdf)

hcouplet, 2023-06-04:
Here is another one interesting on the hybrid model.  Setting the backflow power.
This is not in the documentation, but  a guy discovered it (see https://diysolarforum.com/threads/solis-s5-various-issues.52383/#post-674717)

I tested it and confirm it's working on my inverter.

  - platform: modbus_controller
    modbus_controller_id: modbus_master
    id: set_backflow_power
    name: "Set Backflow Power"
    register_type: holding
    address: 43074
    value_type: U_WORD
    unit_of_measurement: 'W'
    icon: mdi:transmission-tower-export
    entity_category: config
    mode: box
    multiply: 0.01

hn, 2023-06-21:
@hcouplet I added backflow power setting with https://github.com/hn/ginlong-solis/commit/33a4d838f648c4941585b6bdbf764e5ea1325876 . Can you please double check, I slightly modified your code:
- add min value
- changed icon to `tower-import`, export from inverter is import to the grid, right?

closed 2023-06-21 by hn