hn/ginlong-solis #10: Backup the stock firmware fails

Belaial, 2023-06-12:
Trying to backup the stock firmware fails for me, to the best of my knowledge I have followed the instructions so need to ask for help...

I am quite sure I am able to "Set the MCU to UART boot mode (https://github.com/hn/ginlong-solis/issues/9)" since the stick does not start flashing all it's LEDs, only power lit, if the stick boots regularly all the LEDs start to flash.

Once in the mode where only the power LED is lit I try to run

`./rtltool.py -p /dev/ttyUSB0 rf 0x8000000 0x800000 solis-s3-firmware-1012f.bin`

But I get the following error

`./rtltool.py -p /dev/ttyUSB0 rf 0x8000000 0x800000 solis-s3-firmware-1012f.bin
  File "/home/test/Desktop/solis/./rtltool.py", line 56
    print 'Error: Open %s, %d baud!' % (port, baud)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(...)?
`

I am using the rtltool linked in the guide ( https://github.com/pvvx/RTL0B_SDK/blob/master/mkb/rtltool.py )

Any idea what I could be doing wrong? I have also tried with the stick not connected and the same error appears so I guess there is something wrong with my rtltool.py or the arguments I am trying to pass.

hn, 2023-06-12:
Try using python2, e.g. `python2 ./rtltool.py ...`

Belaial, 2023-06-12:
Forgot tom test that.... read about that here https://github.com/hn/ginlong-solis/issues/2 now when you mentioned it.

New error,

`python2 ./rtltool.py -p /dev/ttyUSB0 gf
Traceback (most recent call last):
  File "./rtltool.py", line 9, in <module>
    import serial
ImportError: No module named serial`

Will try and Google that one also, maybe I can solve something myself :)

closed 2023-06-12 by Belaial