hn/ginlong-solis #11: Updating firmware

jonnyrider, 2023-06-12:
Sorry for the very basic question, but even though I'm familiar with ESPHome and Home Assistant, I've never actually flashed the firmware on any other device!

My understanding is that I need to connect the S3 wifi stick via a serial adapter to my Raspberry Pi and then I can run the commands in the readme. Will the following adapter work?

https://www.amazon.co.uk/dp/B08CRL6KHV?psc=1&ref=ppx_yo2ov_dt_b_product_details

Do I then attach wires between the stick and the serial adapter (TX to TX, RX to RX and GND to GND)?

[image]

I understand that I will need to connect the TX pin to GND at boot so the firmware can be flashed.

I then think I just type the code into a terminal on the Pi and the stick will be updated with ESPHome firmware that will provide data directly to Home Assistant?

Thanks in advance for any help, I don't want to fry the wifi stick by doing this wrong!

hn, 2023-06-12:
Everything you write is correct in principle (you just have to cross RX-TX lines, but that's a detail). But beware, this is a hackish project in its early stages, I can not guarantee anything). But I think you will not fry the stick.

Please also see #10.

For the serial adapter, I think a [cheaper one](https://www.amazon.co.uk/dp/B09F6GZGX6/) should do the job better.

jonnyrider, 2023-06-12:
Thanks for the quick response!

I was originally going to create my own stick following this: https://github.com/incub77/solis2mqtt which is why I ordered that serial adapter. I will give it a go with that first (it arrives later) and if it doesn't work I'll try the adapter you mentioned.

closed 2023-06-12 by jonnyrider

jonnyrider, 2023-06-15:
I tried this last night, but couldn't get the device to boot into the flash memory. I think I've got the wires round the wrong way, can you please help with where each wire should go when connected to the below?

[image]

Thanks!

hn, 2023-06-16:
I think you can not use this adapter. You need a 5V pin to power the stick (VCC).

hn, 2023-06-17:
Please also see https://github.com/hn/ginlong-solis/issues/9#issuecomment-1595643051

jonnyrider, 2023-06-18:
> Please also see #9 (comment)

Thanks, I've bought a new usb stick for this, but it only seems to boot up
the stick randomly. Probably something I'm doing, so will keep looking into
this!