hn/ginlong-solis #12: Stick won't connect back to WiFi

Belaial, 2023-07-08:
First of I found this issue https://github.com/kuba2k2/libretiny/issues/113 but that seems to be resolved or at least closed so not sure if this is a new issue or the same.

Regarding versions and so on running on the stick I just got my stick to work last night (07 July 2023) so everything is the latest there is to my knowledge following the guide here https://github.com/hn/ginlong-solis#replacing-the-main-application

The stick worked perfect last night for a few hours when I was testing things out, this morning when I woke up everything was still working, the stick goes offline during the night on my inverter but that is expected since the inverter powers down the COM-port, I know this from running this project https://github.com/incub77/solis2mqtt before getting my stick re-flashed, I had to use external power on the Pi to keep it running during the night.

This morning I tested more things, when I powered down a access point I have in the house all of a sudden I lost the Solis stick, I waited a few minutes expecting it to reconnect back but it did not, I powered on the access point again and waited but the stick did not reconnect. I have now power cycled the stick 4 times, waited 10-20 minutes in between but it has not reconnected.

I don't have easy serial access to the stick right now since I assembled the case again but I guess that is the next step to see what is going on unless someone have some other tips for me.

The only thing that worries me is this line I noticed during boot of the Solis stick

`[W][ota:102]: Last Boot was an unhandled reset, will proceed to safe mode in 7 restarts`

I saw this number go down every time I booted the stick with serial connected so.... maybe the stick is stuck in safe mode now?

..... Now all of a sudden it just works again.....

So.... during the time I was writing this ticket (20 minutes maybe since last stick power cycle) the stick just all of a sudden connected back after 2927 seconds (had a rolling ping towards it that is how I know the number of second, it was rebooted several times during these 2927 seconds so why it decided to work now I have no clue...)

I will post this ticket anyway now in case there is some discussion to have regarding this, if this is excepted behavior I guess we can just close this ticket.

hn, 2023-07-09:
A fundamental WiFi problem (which caused WiFi not to work at all) was fixed [here](https://github.com/hn/ginlong-solis/issues/2#issuecomment-1531514620).

The "WiFi reconnection problem" still exists, at least to my knowledge (my WiFi is stable and therefore I had no problems). As far as I understand, this [is caused by the RealTek SDK](https://github.com/kuba2k2/libretiny/issues/113#issuecomment-1537481527) and can not be fixed within this project.

Maybe it's a good idea to implement some kind of ESPhome `time trigger` which resets the stick after not beeing able to connect via WiFi. I don't know :)

closed 2023-07-09 by hn

Belaial, 2023-07-09:
Ah ok, I see.

I know the WiFi network as a whole is not the issue since I have other WiFi devices in the same area as the Solis stick, those devices have 100% uptime for many days in a row.

This is a ping monitoring in Home Assistant towards the Solis stick

[image]

The large gray area between 00:30-04:00 can be ignored, that is when my inverter powers down the COM-port leaving the stick without power, therefore also a long period when it's offline, the smaller gray gaps are when the stick is randomly drops connection going offline for XX amount of minutes.

I have experimented some with moving a access point closer / further away from the Solis stick but it does not help, the issue looks to be the same at -55dBm as it does now on -75dBm

[image]

My data collection is very short still so hard to say how it would look over a longer period of time.