hn/ginlong-solis #16: Compilation fails with latest LibreTiny

Belaial, 2023-07-17:
Just as a test I tried to compile the firmware with the latest version (when writing this - [v1.2.0](https://github.com/kuba2k2/libretiny/releases/tag/v1.2.0))  of [LibreTiny](https://github.com/kuba2k2/libretiny) and got the following error

```
python3 -m esphome compile solis-esphome-emw3080.yaml
INFO ESPHome 2023.8.0-dev
INFO Reading configuration solis-esphome-emw3080.yaml...
Failed config

libretiny: [source solis-esphome-emw3080.yaml:44]
  
  The LibreTiny component is now split between supported chip families.
  Migrate your config file to include a chip-based configuration, instead of the 'libretiny:' block.
  For example 'bk72xx:' or 'rtl87xx:'.
  board: generic-rtl8710bx-4mb-980k
  framework: 
    version: latest
```

I guess some modifications are needed to make this project compatible with latest [LibreTiny](https://github.com/kuba2k2/libretiny).

hn, 2023-07-17:
Thanks for reporting. Looks like it is as easy as just changing one line:

```
-libretiny:
+rtl87xx:
```

I'll check that later.

Belaial, 2023-07-17:
> Thanks for reporting. Looks like it is as easy as just changing one line:
> 
> ```
> -libretiny:
> +rtl87xx:
> ```
> 
> I'll check that later.

I tested this and it now compiled without issues!
I would do a pull request but I have never done that and have no clue how to do that correctly so just reporting here that it works.

commit a3350517 (2023-07-17): fix for LibreTiny 1.2.0, closes #16
closed 2023-07-17 by hn