hn/ginlong-solis #17: Commit b25f978 prohibiting reboot if WiFi is not connected?

Belaial, 2023-07-25:
Hi, last night I switched out my Raspberry Pi running https://github.com/incub77/solis2mqtt for the modified Solis stick, I waited until after midnight since I did not want to mess up my graphs in HA.

This morning I woke up hoping to see everything work (Inverter had the COM-port down when I plugged in the stick so it did not boot up until later when I was asleep) to my surprise tho the stick was offline when I woke up, I went out to check on it and it had power but no connection, I pressed the reset button on the back of the stick and it then booted up and connected just fine.

I later did some quick tests with this complied and flashed (used 3 minutes to put it under the `interval: 280s` in commit https://github.com/hn/ginlong-solis/commit/b25f9782a94c28bd399e897d7c897bb8f108f744,

```
wifi:
  reboot_timeout: 3min
```

But it seems the stick does not reboot when WiFi is not connected, could this somehow be related to the commit https://github.com/hn/ginlong-solis/commit/b25f9782a94c28bd399e897d7c897bb8f108f744 which fixed the hardware watchdog we had issues with before?

I was asked to check this before (see below), I did however not test what would happen in WiFi was not connected, only that it stopped the reboot every 15 minutes due to the hardware watchdog.

> @Belaial Can you please check [b25f978](https://github.com/hn/ginlong-solis/commit/b25f9782a94c28bd399e897d7c897bb8f108f744) . It sends a 100ms 'high' pulse every just under 5 minutes to reset the WDT, but only if WiFi is connected (so it should catch WiFi reconnect problems as well).

I only have one Solis S3 stick and it's now connected to the inverter so don't want to remove it since it's collecting data (can remove it after midnight if needed for more testing), do you happen to have more than one stick @hn ? Was thinking if you could do a quick test to see if https://github.com/hn/ginlong-solis/commit/b25f9782a94c28bd399e897d7c897bb8f108f744 interferes with the reboot if WiFi is not connected.

I don't see how https://github.com/hn/ginlong-solis/commit/b25f9782a94c28bd399e897d7c897bb8f108f744 could interfere with the reboot if not connected to WiFi but need to ask since I'm out of ideas..

hn, 2023-07-28:
I recommend to remove all timeouts etc. If no API client is connected for 15 minutes, ESPhome will reboot the device anyway.

I'm currently thinking about resetting the WDT only if there is active ModBus traffic *and* API clients connected, something like this:

```
--- a/solis-esphome-emw3080.yaml
+++ b/solis-esphome-emw3080.yaml
@@ -87,15 +87,6 @@
               switch.is_on: led_green_net
             then:
               - switch.turn_off: led_green_net
-  - interval: 280s
-    then:
-      if:
-        condition:
-          wifi.connected:
-        then:
-          - switch.turn_on: wdt_reset
-          - delay: 100ms
-          - switch.turn_off: wdt_reset
 
 script:
   - mode: restart
@@ -104,6 +95,13 @@
       - switch.turn_on: led_orange_com
       - delay: 500ms
       - switch.turn_off: led_orange_com
+      - if:
+          condition:
+            api.connected:
+          then:
+            - switch.turn_on: wdt_reset
+            - delay: 100ms
+            - switch.turn_off: wdt_reset
 
 uart:
   id: modbus_uart
```

Please do not expect any significant updates from me during the next 2-3 weeks.

Belaial, 2023-07-29:
Sounds like a good plan, any way to also detect `CRC Check failed!` and in return reboot when that happens? (somewhat off topic and should probably be asked here https://github.com/hn/ginlong-solis/issues/4)

> Please do not expect any significant updates from me during the next 2-3 weeks.

Ok no worries, I have a quite OK working setup atm with the solution I came up with here https://github.com/hn/ginlong-solis/issues/4#issuecomment-1653569192

Belaial, 2023-07-31:
Just a small update regarding this ticket, with my HA automation that reboots the stick if the `CRC Check failed!`, that has now occurred 18 times in the past 5 days. Out of these 18 restarts the stick has returned instantly 16 times and only 2 times has it been offline for a longer period, however I guess that is due to the already known WiFi instability that has already been reported and discusses in other tickets.

So, with that information on hand I assume the https://github.com/hn/ginlong-solis/commit/b25f9782a94c28bd399e897d7c897bb8f108f744 commit does not prohibit rebooting as I first suspected.

If you wanna close this ticket since I seem to have proved myself wrong that is fine, but I guess you might wanna keep it open due to the changes you have planned here https://github.com/hn/ginlong-solis/issues/17#issuecomment-1655942034

commit ea1883a1 (2023-08-12): reset hw wdt if ModBus traffic and HA api both are ok, closes #17
closed 2023-08-12 by hn