hn/ginlong-solis #18: Can't build firmware with latest instructions

Belaial, 2023-08-15:
Hi,

I was going to build the latest firmware to include the fix for modbus getting desynchronised, however I must be doing something wrong.

I started of fresh from the beginning, all works until this line

`wget https://raw.githubusercontent.com/hn/ginlong-solis/libretiny-ringbuffer-workaround.diff`

That throws a `400: Invalid request`, was fixed with using this link instead for me (notice the `/master/` part in the working link)

`https://raw.githubusercontent.com/hn/ginlong-solis/master/libretiny-ringbuffer-workaround.diff`

However later on in the guide when I am about to apply the fix contained in `libretiny-ringbuffer-workaround.diff` I get the following

```

~/.platformio$ patch -p1 < libretiny-ringbuffer-workaround.diff
can't find file to patch at input line 6
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|Workaround for the Arduino ringbuffer, so that it can no longer have a permanently
|inconsistent state. Implemented by removing the redunant storage of the buffer
|state in three variables.
|--- a/packages/framework-arduino-api/api/RingBuffer.h	2023-08-12 13:07:59.074806708 +0200
|+++ b/packages/framework-arduino-api/api/RingBuffer.h	2023-08-11 15:35:11.847630525 +0200
--------------------------
File to patch:
```
I tried looking for the file 

[image]

I also tried to manually find the file but I don't even have the framework-arduino-api folder
```
~/.platformio/packages$ ll
total 44
drwxrwxr-x 11 user user 4096 Aug 15 09:32  ./
drwxrwxr-x  5 user user 4096 Aug 15 09:36  ../
drwx------  7 user user 4096 Aug 15 09:31  library-flashdb/
drwx------  4 user user 4096 Jul  7 22:14  library-freertos/
drwx------  7 user user 4096 Aug 15 09:31  library-freertos-port/
drwx------  7 user user 4096 Jul  7 22:14  library-lwip/
drwx------  6 user user 4096 Aug 15 09:31  library-printf/
drwx------  6 user user 4096 Aug 15 09:31  library-uf2ota/
drwx------  8 user user 4096 Aug 15 09:32  toolchain-gccarmnoneeabi/
drwx------ 10 user user 4096 Jul 17 12:00 'toolchain-gccarmnoneeabi@1.100301.220327'/
drwx------  3 user user 4096 Jul  7 22:14  tool-scons/
```

Am I missing something here? Sorry in advance if I am doing something obviously wrong...

hn, 2023-08-15:
Try compiling the yaml file (this will install dependencies = framework-arduino-api), then the patch file, and finally compiling again.

Belaial, 2023-08-15:
That indeed solved my issue! Thanks @hn for quick support once again! 
Might be good to mention this in the guide for `Replacing the main application`, unless I did something really wrong.

commit 7b674240 (2023-08-15): fix patch order, closes #18
closed 2023-08-15 by Belaial