hn/ginlong-solis #20: Synchronisation intervl

jolamouch, 2023-11-07:
Do you think it would be possible to reduce the data synchronisation interval? At the moment the data is updated every minute. I'd like to go down to 2 seconds for certain sensors. Do you have any ideas?
Thanks

hn, 2023-11-07:
Just be careful not to overload both the ModBus and the inverter. But I don't think that's a problem with 2 seconds if you don't query too many values. In other words, you have to set the `update_interval `for the `modbus_controller` to `2s` and `skip_updates` to a high value for all sensors that are not absolutely necessary (if you use the ESPhome variant)

jolamouch, 2023-11-07:
OK! Thanks for your advice, I understand the reasoning. Maybe I won't reduce the synchronisation frequency too much. Thank you for your responsiveness! And thanks also for your github project, I've been able, with a few mishaps, to get feedback from my inverter data. 
Have a nice evening!

⁣Jonathan P. ​

hn, 2023-11-08:
Thanks. Please support projects like this by reporting both positive and negative experiences back to the community.

closed 2023-11-08 by hn

jolamouch, 2023-11-21:
Hello,
I followed your advice and everything is working perfectly. There are only 3 sensors synchronised at 2s. I'll reset everything to 60s by default when I've finished analysing the tests, to avoid overloading the data and the inverter.
Thanks a lot!

hn, 2023-11-22:
If everything runs stable all the time, you can leave the setting at 2s, I don't think the inverter will wear out.
Just out of interest, are you using an ESP8266 or a Solis stick?

jolamouch, 2023-11-22:
OK I'll leave it on 2s for now. I use an esp32 D1 mini connected directly to the stick terminals. So I have both. I'm thinking of using a hub bus to avoid data conflicts between the two. I'll keep you posted.

jolamouch, 2023-11-29:
Hello,
After a few tests, it doesn't work... I get the following error message:
 
[18:25:08][W][modbus_controller:136]: Duplicate modbus command found: type=0x4 address=33092 count=3
[18:25:08][W][modbus_controller:136]: Duplicate modbus command found: type=0x4 address=33135 count=1
etc...
I guess the update frequency is too short but I have the impression that it worked for a while and not anymore...
I'm continuing my tests.

Translated with www.DeepL.com/Translator (free version)

hn, 2023-11-29:
If you use an ESP and the Solis stick at the same time, it is no wonder that there are interfering (double) packets on the bus, because of course both devices send commands (in your case one device even sends at a high rate).

jolamouch, 2023-11-30:
Hello,
Yes, I understand your reasoning and you're probably right. I've seen other github projects with a hub bus (master-slave bus). Except that I'm getting these error messages even though I haven't yet connected the ESP to the solis stick. I've connected the ESP to my PC to do the tests, and I can't manage to get it below 35s otherwise I get the errors.

jolamouch, 2023-11-30:
Hello,
After a few more tests, it works! Running the tests without connecting the esp to the solis causes the errors. After reconnecting the esp to the solis with an update value of 2s for only 3 sensors (the rest at 60s), the data is correct and there are no bugs in the logs. For the moment ;-)

johndeere6110, 2025-07-08:
Can you help me to get this done, I can't do it?