hn/ginlong-solis #21: Problems with wemos/esphome & Solis RHI-5K-48ES-5G

ErwinP, 2024-02-08:
Hi,

First of all: thank you for making this possible, and for all the work that you have put in this! I don't like the need of pulling my data from the ginlong servers, and this projects would fit all my needs.

I've set up a wemos D1 device using ESPHome as indicated in the readme and the RX/TX lights of the RS485-to-serial adapter are blinking. However, I can't get this to work: the state of all sensors is "NA" and I'm getting the following from the logs:

`[D][modbus_controller:040]: Modbus command to device=1 register=0x8119 countdown=0 no response received - removed from send queue
[D][modbus_controller:040]: Modbus command to device=1 register=0x8121 countdown=0 no response received - removed from send queue
[D][modbus_controller:040]: Modbus command to device=1 register=0x8131 countdown=0 no response received - removed from send queue
[D][modbus_controller:040]: Modbus command to device=1 register=0x8137 countdown=0 no response received - removed from send queue
[D][modbus_controller:040]: Modbus command to device=1 register=0x8161 countdown=0 no response received - removed from send queue`

I'll be very honest to start off: this whole modbus thing is super new to me and I'm still quite confused about this. Any hints as to what is happening or where to start troubleshooting, would be very much appreciated :)

Thank you in advance!

Kind regards,
Erwin

hn, 2024-02-08:
Well, if TX/RX are blinking, RS485 data _likely_ is flowing, but ... you never know.

- Try swapping RX/TX from D1 to adapter. Some of those cheap adapters require RX-RX and TX-TX, some RX-TX crossed.
- Double-check that the inverter itself has ModBus address 1, you can set the address via the LCD screen. If the inverter has an address of e.g. 2, then obviously things like `Modbus command to device=1` will not work.

ErwinP, 2024-02-09:
Hi, thanks for this fast answer!

With respect to the Modbus address, is there any guidance where to set this? Under the advanced settings > RS485 I only see

- Baud Rate: 9600
- Parity: None
- Data bits: 8
- Stop bit: 1

Thanks!
Erwin

ErwinP, 2024-02-29:
I just realized I never gave feedback on this one, sorry!
Actually, when trying to play around with the TX/RX wire, I lost all connection. I therefore suspect that my soldering skills are not as good as I wished for :) I decided to go for the wifi stick, of which the view has a slightly higher wife acceptance factor anyway. But it's giving me headaches anyhow, will open a new issue for that.

Thank you for the feedback!

closed 2024-02-29 by ErwinP