hn/ginlong-solis #22: Tested to reflash Solis stick with ESPHome... I should not have done that

Belaial, 2024-02-22:
Hi again, long time no see :)

So I read that there was a new way to flash the stick via ESPHome, figured why not.... and this is probably where I made a mistake.

I read up on all instructions and ignored the manual flashing part, since my stick was already running custom firmware and was reachable via IP I figured I could just flash the stick via OTA directly instead of doing it manually via serial first.

I put together a configuration, re-checked everything and figured it was all good, validated the config and hit install via wireless, all looked good but when a few minutes passed I figured something was wrong.... I took the stick inside to see what is was doing and it's just boot looping with 

```
ROM:[V0.1]                                                                   
FLASHRATE:4                                                                  
BOOT TYPE:0 XTAL:40000000                                                    
IMG1 DATA[1128:10002000]                                                     
IMG1 ENTRY[800053d:100021ef]                                                 
IMG1 ENTER                                                                   
CHIPID[000000ff]                                                             
read_mode idx:2, flash_speed idx:2                                           
calibration_result:[1:19:13][3:15]                                           
calibration_result:[2:21:11][1:15]                                           
calibration_result:[3:0:0][ff:ff]                                            
calibration_ok:[2:21:11]                                                     
FLASH CALIB[NEW OK]                                                          
OTA2 ADDR[8100000]
OTAx SELE[f8000000]
OTA2 USE
OTA2 SIGN[35393138:31313738]
IMG2 DATA[0x81a80c0:4496:0x10005000]
IMG2 SIGN[RTKWin(10005008)]
IMG2 ENTRY[0x10005000:0x81361ad]
BOOT_FLASH_RDP RDP enable 
RDP bin decryption Failed!
checksum_ipsec = 0x4e784fc0, checksum_rdp_flash = 0xfd90f7a6
System_Init1
System_Init2
I [      0.000] LibreTiny v1.5.0 on generic-rtl8710bx-4mb-980k, compiled at Feb 22 2024 18:08:06, GCC 10.3.1 (-Os)
[I][logger:403]: Log initialized
[C][ota:483]: There have been 1 suspected unsuccessful boot attempts.
[D][lt.preferences:104]: Saving 1 preferences to flash...
[D][lt.preferences:132]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[C][uart.lt:049]: Setting up UART...
[C][switch.gpio:011]: Setting up GPIO Switch 'led_orange_com'...
[D][switch:016]: 'led_orange_com' Turning OFF.
[D][switch:055]: 'led_orange_com': Sending state OFF
[D][switch:016]: 'led_orange_com' Turning OFF.
[C][switch.gpio:011]: Setting up GPIO Switch 'led_green_net'...
[D][switch:016]: 'led_green_net' Turning OFF.
[D][switch:055]: 'led_green_net': Sending state OFF
[D][switch:016]: 'led_green_net' Turning OFF.
[C][switch.gpio:011]: Setting up GPIO Switch 'wdt_reset'...
[D][switch:016]: 'wdt_reset' Turning OFF.
[D][switch:055]: 'wdt_reset': Sending state OFF
[D][switch:016]: 'wdt_reset' Turning OFF.
[D][binary_sensor:034]: 'button_reset': Sending initial state OFF
[C][wifi:038]: Setting up WiFi...
[C][wifi:051]: Starting WiFi...
[C][wifi:052]:   Local MAC: 80:A0:36:A7:F7:1F
interface 0 is initialized
                          interface 1 is initialized

Initializing WIFI ...
WIFI initialized
                [D][wifi:459]: Starting scan...
RTL8195A[HAL]: Hard Fault Error!!!!
RTL8195A[HAL]: R0 = 0x1b9
RTL8195A[HAL]: R1 = 0x1
RTL8195A[HAL]: R2 = 0x0
RTL8195A[HAL]: R3 = 0x9
RTL8195A[HAL]: R12 = 0x1b9
RTL8195A[HAL]: LR = 0x8110bad
RTL8195A[HAL]: PC = 0x1eb8dcf0
RTL8195A[HAL]: PSR = 0x80000000
RTL8195A[HAL]: BFAR = 0xe000ed38
RTL8195A[HAL]: CFSR = 0x20000
RTL8195A[HAL]: HFSR = 0x40000000
RTL8195A[HAL]: DFSR = 0x0
RTL8195A[HAL]: AFSR = 0x0
RTL8195A[HAL]: PriMask 0x0
RTL8195A[HAL]: BasePri 0x0
RTL8195A[HAL]: SVC priority: 0x00
RTL8195A[HAL]: PendSVC priority: 0xf0
RTL8195A[HAL]: Systick priority: 0xf0
```

If I try to pull TX pin 21 low during boot to re-flash the stick via serial it just won't work

```
python2 rtltool2.py -p /dev/ttyUSB0 gf
Connecting...
Flash Status value: 0x15

python2 rtltool2.py -p /dev/ttyUSB0 wf 0xb000 solis-emw3080-0x00B000.bin 
Connecting...
Write Flash data 0x0800b000 to 0x080b424c from file: solis-emw3080-0x00B000.bin ...
Error: Write Flash!
```

I see that the Flash status is 0x15 and not 0x40 as in the guide, quite sure I did not see this before when we did all the troubleshooting back in last year.

Any ideas what could be wrong? Did I finally manage to kill the stick with my curiosity....

hn, 2024-02-22:
Hi Belaial, I also had flash status 0x15 a long time ago ... and when I tried to flash again an hour later, it was magically 0x40 again, I don't know what went wrong. You might want to try ltchiptool for flashing, the latest version worked fine for me (tested just once).

Belaial, 2024-02-22:
Hi, thx for the quick reply!

I have tried multiple times over the past 2 hours, always get 0x15 so far.

Will look into ltchiptool and let you know how it goes.

hn, 2024-02-22:
I strongly suspect that the 0x15 / 0x40 error depends on the power (current) that the USB port can supply.

I think everything is fine with your configuration (except that it is not advisable to paste passwords ;-) ). You might want to recompile / flash the new binary again and see if it just works.

Your ESPhome dashboard uses LibreTiny v1.5.0 (which is good as it contains an important OTA fix). I haven't tested 1.5.0 yet.

Belaial, 2024-02-22:
Hmm, yeah I guess that is a valid theory regarding power / current issues, the "Hard Fault Error!!!!" I saw that on both of the USB-ports (also this was most likely the status in the inverter also since it never connected back after my OTA flash).

The pasted passwords won't be used :smile: the `password: "VerySecretPassword"` was edited from my real password since I did not want to paste that the API key and OTA password will be replaced also :+1: 

I am a bit short on time this weekend so won't have time to test / recompile before maybe Saturday night, otherwise Sunday evening most likely, I will let you know how it goes before we close this ticket down in case there is some real issue, forgot to mention but I have the latest HA updates and ESPHome addon updates so all that should be as good as it can be.

hn, 2024-02-24:
Same issue here with v1.5.0 ( "Starting scan... RTL8195A[HAL]: Hard Fault Error!!!!").

v1.4.1 works fine. I more or less see only documentation and Beken-related changes from 1.4.1 to 1.5.0, so I need to investigate in detail what's wrong there, don't have time right now for that.

@kuba2k2

Belaial, 2024-02-26:
My weekend did unfortunately not include the free time to look at this, glad you verified @hn that you see the same as me.
I was gonna edit the title of this issue to something more fitting but I see you already did that :+1: 

I will follow along here and see when a possible fix might be ready, I am then ready to test again.

hn, 2024-02-29:
Fixed in LT 1.5.1, please also see https://github.com/hn/ginlong-solis/issues/23#issuecomment-1972006075

closed 2024-02-29 by hn

ErwinP, 2024-03-01:
Btw, there is really a strong sensitivity towards the USB port and the power.

I was using an old Surface Pro tablet to flash, which only has one USB port. Still facing the ACK errors I got during flashing, I plugged the charging cable into the tablet. Tried flashing 2 times, and after completing (and still receiving those ACK errors) I got an error at the very end of the flashing procedure indicating a strong failure. After having removed the power source again, flashing completed successfully several times after each other.

Belaial, 2024-03-02:
This morning I took the "risk" again :)

I had to manually edit LT version to ```version: 1.5.1``` but I guess that was expected as mentioned in https://github.com/hn/ginlong-solis/issues/23#issuecomment-1972006075 for that not to be needed I guess ESPHome addon (or something ESPHome related) needs to be updated on their end.

Either way OTA and everything worked just fine! MANY THANK @hn for this wonderful project! And also great work everyone involved in solving this! I hope I don't need to be the next one to find a bug :smile: (ofc there won't be any since this project is perfect :wink: )

ErwinP, 2024-03-02:
Can't agree more, thank you @hn to make this possible and help with all the troubleshooting and development. **Very** much appreciated, really!

Belaial, 2024-04-17:
FYI

https://esphome.io/changelog/2024.4.0.html

"Bump LibreTiny version to 1.5.1 [esphome#6500](https://github.com/esphome/esphome/pull/6500) by [@kuba2k2](https://github.com/kuba2k2)"

I changed out ```version: 1.5.1``` to ```version: latest``` and re-flashed, all worked great!

Been running for 2+ hours now without issues!