hn/ginlong-solis #24: Component modbus_controller took a long time for an operation

Belaial, 2024-03-02:
This is maybe more of a question than an issue I guess but creating one anyway to make sure.

This morning I flashed the latest possible build with ESPHome and LibreTiny 1.5.1, all works perfect so far for 10 minutes.

Since I now have the stick compiled and monitored via Home Assistant I saw the log output after compile and OTA update and I then noticed this between the readouts of data from the inverter.

```
[07:57:27][W][component:214]: Component modbus_controller took a long time for an operation (0.12 s).
[07:57:27][W][component:215]: Components should block for at most 20-30ms.

[07:58:25][W][component:214]: Component modbus_controller took a long time for an operation (0.06 s).
[07:58:25][W][component:215]: Components should block for at most 20-30ms.

[07:58:25][W][component:214]: Component modbus_controller took a long time for an operation (0.08 s).
[07:58:25][W][component:215]: Components should block for at most 20-30ms.

[07:58:25][W][component:214]: Component modbus_controller took a long time for an operation (0.12 s).
[07:58:25][W][component:215]: Components should block for at most 20-30ms.
```

If this is expected warnings just ignore and close this ticket but maybe there is something to tweak to avoid these warnings in the logs?

hn, 2024-03-02:
This is an ESPHome discussion and not related to this project, see https://github.com/esphome/issues/issues/4717

closed 2024-03-02 by hn

Belaial, 2024-03-02:
Ok, thanks for the clarification @hn