hn/ginlong-solis #25: no success with ESPHome and Wemos D1 mini

dkotte, 2024-03-06:
I try the Wemos D1 mini - variant and got the following output:

`INFO ESPHome 2024.2.1
INFO Reading configuration /config/esphome/esphome-web-53bff9.yaml...
INFO Starting log output from fhem/sensors/ESPHome-Solis/debug
INFO Connected to MQTT broker!
[08:40:47][E][modbus_controller:095]: Modbus error - last command: function code=0x4  register address = 0x8161  registers count=1 payload size=0
[08:41:13][V][sensor:043]: 'Uptime Sensor': Received new state 104.954002
[08:41:16][V][modbus_controller:181]: Updating modbus component
[08:41:16][V][modbus_controller:147]: Range : 80E9 Size: 1 (4) skip: 9
[08:41:16][V][modbus_controller:147]: Range : 80E8 Size: 1 (4) skip: 9
[08:41:16][V][modbus_controller:147]: Range : 8107 Size: 6 (4) skip: 9
[08:41:16][V][modbus_controller:147]: Range : 8121 Size: 2 (4) skip: 0
[08:41:16][V][modbus_controller:147]: Range : 8137 Size: 2 (4) skip: 0
[08:41:16][V][modbus_controller:147]: Range : 8161 Size: 1 (4) skip: 0
[08:41:16][V][modbus_controller:147]: Range : 818D Size: 2 (4) skip: 9
[08:41:16][W][component:214]: Component modbus_controller took a long time for an operation (0.10 s).
[08:41:16][V][modbus_controller:046]: Sending next modbus command to device 1 register 0x8119 count 4
[08:41:16][V][modbus_controller:509]: Command sent 4 0x8119 4
[08:41:16][V][modbus:042]: Modbus received Byte  1 (0X1)
[08:41:16][V][modbus:042]: Modbus received Byte  3 (0X3)
[08:41:16][E][modbus_controller:091]: Modbus error function code: 0x4 exception: 5 
[08:41:16][E][modbus_controller:095]: Modbus error - last command: function code=0x4  register address = 0x8119  registers count=4 payload size=0
[08:41:16][V][modbus:199]: Modbus write: 01.04.81.21.00.02.09.FD (8)
[08:41:17][V][modbus:042]: Modbus received Byte  1 (0X1)
[08:41:17][D][modbus:119]: Modbus error function code: 0x84 exception: 5
[08:41:17][E][modbus_controller:095]: Modbus error - last command: function code=0x4  register address = 0x8121  registers count=2 payload size=0
[08:41:17][V][modbus_controller:509]: Command sent 4 0x8131 1
[08:41:17][V][modbus:042]: Modbus received Byte  1 (0X1)
[08:41:17][V][modbus:042]: Modbus received Byte  131 (0X83)
[08:41:17][E][modbus_controller:091]: Modbus error function code: 0x4 exception: 5 
[08:41:17][V][modbus:199]: Modbus write: 01.04.81.37.00.02.E8.39 (8)
[08:41:17][V][modbus:042]: Modbus received Byte  1 (0X1)
[08:41:17][V][modbus:042]: Modbus received Byte  131 (0X83)
[08:41:17][D][modbus:119]: Modbus error function code: 0x84 exception: 5
[08:41:17][E][modbus_controller:095]: Modbus error - last command: function code=0x4  register address = 0x8137  registers count=2 payload size=0
[08:41:17][V][modbus:042]: Modbus received Byte  1 (0X1)
[08:41:17][V][modbus:042]: Modbus received Byte  3 (0X3)
[08:41:17][E][modbus_controller:091]: Modbus error function code: 0x4 exception: 5 
[08:41:17][E][modbus_controller:095]: Modbus error - last command: function code=0x4  register address = 0x8161  registers count=1 payload size=0
[08:41:23][V][sensor:043]: 'WiFi Signal': Received new state -72.000000
[08:42:13][V][sensor:043]: 'Uptime Sensor': Received new state 164.953995
[08:42:16][V][modbus_controller:181]: Updating modbus component
[08:42:16][V][modbus_controller:147]: Range : 8131 Size: 1 (4) skip: 0
[08:42:16][V][modbus_controller:147]: Range : 8144 Size: 3 (4) skip: 8
[08:42:16][V][modbus_controller:147]: Range : 8173 Size: 1 (4) skip: 8
[08:42:16][V][modbus_controller:046]: Sending next modbus command to device 1 register 0x8119 count 4
[08:42:16][V][modbus:042]: Modbus received Byte  1 (0X1)
[08:42:16][E][modbus_controller:095]: Modbus error - last command: function code=0x4  register address = 0x8119  registers count=4 payload size=0
[08:42:16][V][modbus:042]: Modbus received Byte  1 (0X1)
[08:42:17][V][modbus:042]: Modbus received Byte  131 (0X83)
[08:42:17][D][modbus:119]: Modbus error function code: 0x84 exception: 5
[08:42:17][E][modbus_controller:095]: Modbus error - last command: function code=0x4  register address = 0x8121  registers count=2 payload size=0
[08:42:17][V][modbus_controller:509]: Command sent 4 0x8131 1
[08:42:17][V][modbus:042]: Modbus received Byte  1 (0X1)
[08:42:17][V][modbus:042]: Modbus received Byte  3 (0X3)
[08:42:17][E][modbus_controller:091]: Modbus error function code: 0x4 exception: 5 
[08:42:17][V][modbus:199]: Modbus write: 01.04.81.37.00.02.E8.39 (8)
[08:42:17][V][modbus:042]: Modbus received Byte  1 (0X1)
[08:42:17][V][modbus:042]: Modbus received Byte  131 (0X83)
[08:42:17][D][modbus:119]: Modbus error function code: 0x84 exception: 5
[08:42:17][E][modbus_controller:095]: Modbus error - last command: function code=0x4  register address = 0x8137  registers count=2 payload size=0
[08:42:17][V][modbus:199]: Modbus write: 01.04.81.61.00.01.48.28 (8)
[08:42:17][V][modbus:042]: Modbus received Byte  1 (0X1)
[08:42:17][V][modbus:042]: Modbus received Byte  131 (0X83)
[08:42:17][D][modbus:119]: Modbus error function code: 0x84 exception: 5
[08:42:17][E][modbus_controller:095]: Modbus error - last command: function code=0x4  register address = 0x8161  registers count=1 payload size=0
[08:43:13][V][sensor:043]: 'Uptime Sensor': Received new state 224.966003
[08:43:16][V][modbus_controller:181]: Updating modbus component
[08:43:16][V][modbus_controller:147]: Range : 80EC Size: 4 (4) skip: 7
[... 91 lines cut ...]
[08:46:16][V][modbus:199]: Modbus write: 01.04.81.19.00.04.08.32 (8)
[08:46:16][V][modbus:042]: Modbus received Byte  1 (0X1)
[08:46:16][V][modbus:042]: Modbus received Byte  193 (0Xc1)
[08:46:16][E][modbus_controller:091]: Modbus error function code: 0x4 exception: 2 
[08:46:16][E][modbus_controller:095]: Modbus error - last command: function code=0x4  register address = 0x8119  registers count=4 payload size=0
[08:46:16][V][modbus:199]: Modbus write: 01.04.81.21.00.02.09.FD (8)
[08:46:16][V][modbus:042]: Modbus received Byte  1 (0X1)
[08:46:17][E][modbus_controller:095]: Modbus error - last command: function code=0x4  register address = 0x8121  registers count=2 payload size=0
[08:46:17][V][modbus:199]: Modbus write: 01.04.81.31.00.01.48.39 (8)
[08:46:17][V][modbus:042]: Modbus received Byte  1 (0X1)
[08:46:17][V][modbus:042]: Modbus received Byte  194 (0Xc2)
[08:46:17][D][modbus:119]: Modbus error function code: 0x84 exception: 2
[08:46:17][E][modbus_controller:095]: Modbus error - last command: function code=0x4  register address = 0x8131  registers count=1 payload size=0
[08:46:17][V][modbus:199]: Modbus write: 01.04.81.37.00.02.E8.39 (8)
[08:46:17][V][modbus:042]: Modbus received Byte  1 (0X1)
[08:46:17][V][modbus:042]: Modbus received Byte  194 (0Xc2)
[08:46:17][D][modbus:119]: Modbus error function code: 0x84 exception: 2
[08:46:17][E][modbus_controller:095]: Modbus error - last command: function code=0x4  register address = 0x8137  registers count=2 payload size=0
[08:46:17][V][modbus:199]: Modbus write: 01.04.81.61.00.01.48.28 (8)
[08:46:17][V][modbus:042]: Modbus received Byte  1 (0X1)
[08:46:17][V][modbus:042]: Modbus received Byte  194 (0Xc2)
[08:46:17][D][modbus:119]: Modbus error function code: 0x84 exception: 2
[08:46:17][E][modbus_controller:095]: Modbus error - last command: function code=0x4  register address = 0x8161  registers count=1 payload size=0`

Ideas?

hn, 2024-03-06:
Please use github code markup to format your log, it is currently not possible to understand it.

What inverter do you have and what include file did you use?

dkotte, 2024-03-07:
Behaviour has changed.
Yesterday i got all the data ... but 1 Time only.
Logging (here from remote-logging) shows now the following:
`2024-03-07_17:39:24 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:147]: Range : 8131 Size: 1 (4) skip: 0[0m
2024-03-07_17:39:24 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:147]: Range : 8144 Size: 3 (4) skip: 0[0m
2024-03-07_17:39:24 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:147]: Range : 8173 Size: 1 (4) skip: 0[0m
2024-03-07_17:39:27 MQTT2_new_ESPHome_Solis debug: [0;32m[I][mqtt:274]: MQTT Connected![0m
2024-03-07_17:39:27 MQTT2_new_ESPHome_Solis debug: [0;37m[V][mqtt:373]: subscribe(topic='fhem/sensors/ESPHome-Solis/number/limit_output_power/command')[0m
2024-03-07_17:39:28 MQTT2_new_ESPHome_Solis debug: [0;37m[V][mqtt:373]: subscribe(topic='fhem/sensors/ESPHome-Solis/number/set_backflow_power/command')[0m
2024-03-07_17:39:28 MQTT2_new_ESPHome_Solis debug: [0;37m[V][mqtt:473]: Publish(topic='fhem/sensors/ESPHome-Solis/sensor/uptime_sensor/state' payload='0' retain=1)[0m
2024-03-07_17:39:28 MQTT2_new_ESPHome_Solis debug: [0;37m[V][mqtt:373]: subscribe(topic='fhem/sensors/ESPHome-Solis/button/restart_with_factory_default_settings/command')[0m
2024-03-07_17:39:28 MQTT2_new_ESPHome_Solis debug: [0;37m[V][mqtt:373]: subscribe(topic='fhem/sensors/ESPHome-Solis/button/reboot_device/command')[0m
2024-03-07_17:39:28 MQTT2_new_ESPHome_Solis debug: [0;37m[V][mqtt:373]: subscribe(topic='fhem/sensors/ESPHome-Solis/light/status_led/command')[0m
2024-03-07_17:39:28 MQTT2_new_ESPHome_Solis debug: [0;37m[V][json:038]: Attempting to allocate 512 bytes for JSON serialization[0m
2024-03-07_17:39:28 MQTT2_new_ESPHome_Solis debug: [0;37m[V][json:058]: Size after shrink 48 bytes[0m
2024-03-07_17:39:28 MQTT2_new_ESPHome_Solis debug: [0;37m[V][mqtt:473]: Publish(topic='fhem/sensors/ESPHome-Solis/light/status_led/state' payload='{"color_mode":"onoff","state":"OFF","color":{}}' retain=1)[0m
2024-03-07_17:39:28 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:206]: Register: 0xA82C 1 0 2 offset=0 skip=10 addr=0x3fff4e8c[0m
2024-03-07_17:39:28 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:219]: Started new range[0m
2024-03-07_17:39:28 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:206]: Register: 0xA842 1 0 2 offset=0 skip=10 addr=0x3fff4f8c[0m
2024-03-07_17:39:28 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:278]: Add range 0xA82C 1 skip:10[0m
2024-03-07_17:39:28 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:206]: Register: 0xA842 1 0 2 offset=0 skip=10 addr=0x3fff4f8c[0m
2024-03-07_17:39:28 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:219]: Started new range[0m
2024-03-07_17:39:28 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:206]: Register: 0x80E9 1 0 2 offset=0 skip=10 addr=0x3fff6e90[0m
2024-03-07_17:39:28 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:278]: Add range 0xA842 1 skip:10[0m
2024-03-07_17:39:28 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:206]: Register: 0x80E9 1 0 2 offset=0 skip=10 addr=0x3fff6e90[0m
2024-03-07_17:39:28 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:219]: Started new range[0m
2024-03-07_17:39:28 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:206]: Register: 0x8105 2 0 4 offset=0 skip=10 addr=0x3fff5b24[0m
2024-03-07_17:39:28 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:278]: Add range 0x80E9 1 skip:10[0m
2024-03-07_17:39:28 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:206]: Register: 0x8105 2 0 4 offset=0 skip=10 addr=0x3fff5b24[0m
2024-03-07_17:39:28 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:219]: Started new range[0m
2024-03-07_17:39:29 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:206]: Register: 0x80E8 1 0 2 offset=0 skip=10 addr=0x3fff6dc8[0m
2024-03-07_17:39:29 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:278]: Add range 0x8105 2 skip:10[0m
2024-03-07_17:39:29 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:206]: Register: 0x80E8 1 0 2 offset=0 skip=10 addr=0x3fff6dc8[0m
2024-03-07_17:39:29 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:219]: Started new range[0m
2024-03-07_17:39:29 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:206]: Register: 0x80EC 4 0 2 offset=0 skip=10 addr=0x3fff70c0[0m
2024-03-07_17:39:29 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:278]: Add range 0x80E8 1 skip:10[0m
2024-03-07_17:39:29 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:206]: Register: 0x80EC 4 0 2 offset=0 skip=10 addr=0x3fff70c0[0m
2024-03-07_17:39:29 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:219]: Started new range[0m
2024-03-07_17:39:29 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:206]: Register: 0x8107 2 0 4 offset=0 skip=10 addr=0x3fff5c14[0m
2024-03-07_17:39:30 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:278]: Add range 0x80EC 4 skip:10[0m
2024-03-07_17:39:30 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:206]: Register: 0x8107 2 0 4 offset=0 skip=10 addr=0x3fff5c14[0m
2024-03-07_17:39:30 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:219]: Started new range[0m
[... cut ...]
2024-03-07_17:40:05 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:206]: Register: 0x8137 2 0 4 offset=0 skip=0 addr=0x3fff5884[0m
2024-03-07_17:40:05 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:219]: Started new range[0m
2024-03-07_17:40:05 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:206]: Register: 0x8144 1 0 2 offset=0 skip=10 addr=0x3fff6f98[0m
2024-03-07_17:40:05 MQTT2_new_ESPHome_Solis debug: [0;37m[V][modbus_controller:278]: Add range 0x8137 2 skip:0[0m`

hn, 2024-03-07:
Set log level to default, this ist too much info.

Again: What inverter do you have and what include file did you use?

dkotte, 2024-03-12:
Thanks a lot
First: it works - after soldering a 100 Ohm resistor to the RS485 Adapter and using solis-modbus-inv.yaml File.

History:
i#ve tried both include files (solis-modbus-esinv.yaml and solis-modbus-inv.yaml) without success. After every flashing i got different errors/effects. From Dump over modbus-timeout to a one-time-all-data-send.
I've also tried the "solis2influx.ino" add some breakpoints+serialprintln and see a "mbreadresult " of 5. 
Because my RS485 does not have the R0 (which should enable a 120Ohm resistor)  i soldered a 100Ohm resistor between A+ and B- connector.
Now it started working.