hn/ginlong-solis #27: failed to validate module [usbserial] BTF

TheGroundZero, 2024-04-22:
Hi

I'm attempting to flash my S3-WIFI-ST stick but I can´t even get serial comms to works.

I hooked up the module to my [FTDI adapter](https://www.aliexpress.com/item/32650148276.html).
Set the adapter to 3.3V, tried 5V as well*. ([this](https://github.com/hn/ginlong-solis?tab=readme-ov-file#replacing-the-main-application) says 3V, [this](https://github.com/hn/ginlong-solis/blob/master/solis-wifi-stick-s3-pcb-front.jpg) says 5V?)
* On 5V, green LED turns on and blinks slowly if TX is pulled low, orange blinks rapidly

[image]
VCC -> 3.3V/5V
TX -> RX
RX -> TX
GND -> GND
(Jumper between TX and GND)

**dmesg output**
```
[ 5325.135580] usb 1-4: USB disconnect, device number 10
[ 5364.899209] usb 1-4: new full-speed USB device number 11 using xhci_hcd
[ 5365.321499] usb 1-4: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[ 5365.321506] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5365.321508] usb 1-4: Product: FT232R USB UART
[ 5365.321510] usb 1-4: Manufacturer: FTDI
[ 5365.321512] usb 1-4: SerialNumber: A50285BI
[ 5365.350420] BPF: 	 type_id=138692 bits_offset=0
[ 5365.350425] BPF:  
[ 5365.350426] BPF: Invalid name
[ 5365.350428] BPF: 
[ 5365.350429] failed to validate module [usbserial] BTF: -22
```

Of course ltchuptool can´t identify the correct serial port.

Stick has a MXCHIP EMW3080-E
I'm working on Pop!_OS (`Linux 6.8.0-76060800daily20240311-generic #202403110203~1711393930~22.04~331756a SMP PREEMPT_DYNAMIC Mon M x86_64 x86_64 x86_64 GNU/Linux`)

hn, 2024-04-22:
I always use(d) 3v3. Using 5v may kill your S3 stick, never did that.

You have to temporarily jumper TX of the S3 stick and GND, not TX of the serial adapter. Release TX-GND after some seconds.

"failed to validate module" likely is a general problem with your linux setup, fix that first.

TheGroundZero, 2024-04-22:
After a reboot, the system does seem to find the device.
I assume during the installation of ltchiptool something got installed that wasn't properly initialized until after the reboot?

I was attempting to use my FlipperZero as UART bridge and it also at first didn't want to connect to my system until after the reboot.

```
[ 1532.813531] usb 1-4: new full-speed USB device number 8 using xhci_hcd
[ 1533.236174] usb 1-4: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[ 1533.236181] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1533.236184] usb 1-4: Product: FT232R USB UART
[ 1533.236185] usb 1-4: Manufacturer: FTDI
[ 1533.236187] usb 1-4: SerialNumber: A50285BI
[ 1533.259203] ftdi_sio 1-4:1.0: FTDI USB Serial Device converter detected
[ 1533.259244] usb 1-4: Detected FT232R
[ 1533.267692] usb 1-4: FTDI USB Serial Device converter now attached to ttyUSB0
```

With that out of the way, I'm running against a next issue: timeout while reading.

```
I: Connecting to 'Realtek AmebaZ' on /dev/ttyUSB0 @ 1500000
I: Connect UART2 of the Realtek chip to the USB-TTL adapter:
I: 
I:     --------+        +---------------------
I:          PC |        | RTL8710B            
I:     --------+        +---------------------
I:          RX | ------ | TX2 (Log_TX / PA30) 
I:          TX | ------ | RX2 (Log_RX / PA29) 
I:             |        |                     
I:         GND | ------ | GND                 
I:     --------+        +---------------------
I:  
I: Using a good, stable 3.3V power supply is crucial. Most flashing issues
I: are caused by either voltage drops during intensive flash operations,
I: or bad/loose wires.
I:  
I: The UART adapter's 3.3V power regulator is usually not enough. Instead,
I: a regulated bench power supply, or a linear 1117-type regulator is recommended.
I:  
I: In order to flash the chip, you need to enable download mode.
I: This is done by pulling CEN to GND briefly, while still keeping the TX2 pin
I: connected to GND.
I:  
I: Do this, in order:
I:  - connect CEN to GND
I:  - connect TX2 to GND
I:  - release CEN from GND
I:  - release TX2 from GND

E: TimeoutError: Timeout while linking
E: |-- File "/usr/local/lib/python3.10/dist-packages/ltchiptool/soc/ambz/util/ambztool.py", line 188, in link
```

hn, 2024-04-22:
First Google hit: https://github.com/libretiny-eu/libretiny/issues/221

TheGroundZero, 2024-04-22:
I am pulling TX (pin next to VCC) low by connecting it to GND. Jammed a male-male dupont in the pinheads of the duponts of the TX and GND wires.

Where is CEN exposed?

[image]
[image]
[image]

TheGroundZero, 2024-04-22:
Did some testing with minicom but never really got any input.

Retried with FTDI set to 5V and now at least I got input in Minicom (no jumper between GND and TX)
```
Welcome to minicom 2.8                                                            
                                                                                  
OPTIONS: I18n                                                                     
Port /dev/ttyUSB0, 01:04:48                                                       
                                                                                  
Press CTRL-B Z for help on special keys                                           

CvbbbFCvbFbFbCvbFbFbFCvbFbFsskrSK
FSSSWSGFRSSSSCvbbCvbFWScF2SSSWSGF
SS
```

Removed power from Solis board. connected jumper, powered board, removed jumper after 5s and ran ltchiptool again
```
python3 -m ltchiptool flash info -d /dev/ttyUSB0 RTL8710B
I: Connecting to 'Realtek AmebaZ' on /dev/ttyUSB0 @ 1500000
I: Transmission successful (ACK received).
I: Transmission successful (ACK received).
I: |-- Success! Chip info: RTL8710BN
I: Reading chip info...
I: Chip: RTL8710BN
I: Transmission successful (ACK received).
I: Transmission successful (ACK received).
E: SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
E: |-- File "/home/user/.local/lib/python3.10/site-packages/serial/serialposix.py", line 595, in read
```
Which, I assume, is because minitool was already connected to the serial port

Closing minicom, power cycling the board with the jumper, removing jumper after 5s and running ltchiptool
```
python3 -m ltchiptool flash info -d /dev/ttyUSB0 RTL8710B
I: Connecting to 'Realtek AmebaZ' on /dev/ttyUSB0 @ 1500000
I: Transmission successful (ACK received).
I: Transmission successful (ACK received).
I: |-- Success! Chip info: RTL8710BN
I: Reading chip info...
I: Chip: RTL8710BN
I: Transmission successful (ACK received).
I: Transmission successful (ACK received).
I: +---------------------+--------------------------------+
I: | Name                | Value                          |
I: +---------------------+--------------------------------+
I: | Chip Type           | RTL8710BN                      |
I: | MAC Address         | D0:BA:E4:88:BC:7E              |
I: |                     |                                |
I: | Flash ID            | 68 40 17                       |
I: | Flash Size (real)   | 8 MiB                          |
I: |                     |                                |
I: | OTA2 Address        | 0x8100000                      |
I: | RDP Address         | 0x8087000                      |
I: | RDP Length          | 0xFF0                          |
I: | Flash SPI Mode      | DIO                            |
I: | Flash SPI Speed     | 71MHZ                          |
I: | Flash ID (system)   | 00FF                           |
I: | Flash Size (system) | 2 MiB                          |
I: | LOG UART Baudrate   | 115200                         |
I: |                     |                                |
I: | SYSCFG 0/1/2        | 40000200 / 02010301 / 00000001 |
I: | ROM Version         | V0.1                           |
I: | CUT Version         | 0                              |
I: +---------------------+--------------------------------+
I: |-- Finished in 4.408 s
```

```
python3 -m ltchiptool flash read -d /dev/ttyUSB0 RTL8710B olis-s3-firmware-1012f.bin
I: Connecting to 'Realtek AmebaZ' on /dev/ttyUSB0 @ 1500000
I: Transmission successful (ACK received).
I: Transmission successful (ACK received).
I: |-- Success! Chip info: RTL8710BN
I: Reading Flash (8 MiB) to 'olis-s3-firmware-1012f.bin'
  [################################################################]  100%          I: Transmission successful (ACK received).
I: Transmission successful (ACK received).

I: |-- Finished in 336.155 s
```

TheGroundZero, 2024-04-22:
Setting FTDI to 5V fixed the issue. Perhaps my serial adapter didn't provide 3.3V when set to it so it needed the extra juice of the 5V. Didn't seem to have killed anything.

closed 2024-04-22 by TheGroundZero