hn/ginlong-solis #28: ESPHome missing sensors versus SolisCloud integration

TheGroundZero, 2024-04-23:
I'm (currently) not using many sensors that the inverter provides, but I noticed there is a difference between what I used to get from SolisCloud versus what ESPHome now reports.

SolisCloud / ESPHome
[image][image]

Notably, I appear to be missing the `Solis Battery Power` which I display in one of my dashboards.
Unless that's what `Total DC output power` is supposed to report? It's reporting 0W while I would expect to still be drawing power from my battery as it's still 45% full.

Also, somehow the Solis integration was reporting 2493kWh of Total Energy produced, while ESPHome now "only" reports 2486kWh?
I did, shortly, have a different Solis inverter before it got replaced with my current, so maybe the first only produced 7kWh and the SolisCloud integration is reporting the sum of the 2 although it created 2 devices for the 2 inverters?

TheGroundZero, 2024-04-23:
> Maybe the [ModBus register map](https://github.com/hn/ginlong-solis?tab=readme-ov-file#solis-modbus-register-map-and-rs-485-documentation) helps to understand the meaning of specific values -- or to add more registers. I do not own an ESINV type inverter.

OK, got it. `Total DC output power` is PV production.
[image]

These look interesting for monitoring my battery:
[image]
[image]

hn, 2024-04-25:
The [invert filter](https://esphome.io/components/binary_sensor/index.html#invert) may help.

Please feel free to submit a reasonable set of additional sensors for inclusion into the ESINV yaml when you have completed your tests and you are convinced that the values are relevant to most users.

TheGroundZero, 2024-04-26:
Some of these I disabled in the ESPHome code already (can be nade available on-demand), others I've disabled myself because I don't actively monitor them

[image]

PR #29 (2024-05-30): Add battery sensors to solis-modbus-esinv.yaml
- add battery-related sensors
- refactor some sections
- hide some sensors by default

Fix #28
closed 2024-05-30 by hn