hn/ginlong-solis #32: Error when i compile : need help

helmuteke, 2024-05-17:
Hello

in receive next error when i compile , someone ?,
Someone who can help ?

/data/cache/platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: .pioenvs/solis-emw3080/raw_firmware.ota1.elf section `.ram_heap.data' will not fit in region `BD_RAM'
/data/cache/platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: region `BD_RAM' overflowed by 224 bytes

collect2: error: ld returned 1 exit status

|-- Generated file not found: .pioenvs/solis-emw3080/raw_firmware.ota1.elf
*** [.pioenvs/solis-emw3080/raw_firmware.elf] Error 1

hn, 2024-05-17:
I can reproduce the problem locally with the newest (2024.5.0) ESPHome version. It _seems_ that they changed something so the project does not fit into RAM anymore -- which is strange. This needs further investigation.

If you have the chance to compile the project with some older ESPHome version this might be a solution. As far as I know there is no (easy) way to downgrade the HA add-on.

hn, 2024-05-17:
ESPHome 2024.5.0 might be [broken in a more general aspect](https://github.com/esphome/issues/issues?q=is%3Aissue+is%3Aopen+2024.5.0), people even [suggest to revoke this version](https://github.com/esphome/issues/issues/5809)

helmuteke, 2024-05-17:
so i have to wait i think ?

bandit7311, 2024-05-17:
I

> ESPHome 2024.5.0 might be [broken in a more general aspect](https://github.com/esphome/issues/issues?q=is%3Aissue+is%3Aopen+2024.5.0), people even [suggest to revoke this version](https://github.com/esphome/issues/issues/5809)

I have rolled back my docker to 2024.4.1 and disabled automatic update through Portainer, other people called version 2024.5.0 as a royal PITA

Solong.B

carstengl, 2024-05-17:
Same on my side, I rolled back ESPHome from 2024.5.0 to 2024.4.1 and reinstalled hn's ginlong-solis file on my ESP8266. Afterwards all was working fine again.

helmuteke, 2024-05-17:
Can someone point me how to downgrade , i use esphome in HA

carstengl, 2024-05-17:
> Can someone point me how to downgrade , i use esphome in HA

I am using ESPHome in HA as well and before installing an update for addons HA is creating a new Backup when you have selected the checkbox. 

In "Settings->System->Backups" you should then see a backup called like "addon_xxxxx_esphome_2024.4.1". Click on it, set the checkbox and restore it, afterwards everthing should work again normal. 

[image]

helmuteke, 2024-05-18:
This version was first instll 🙁

hn, 2024-05-20:
They released 2024.5.1. I see no relevant changes for this problem in the changelog and people are still complaining about defunct devices. Did not try to compile solis-emw3080.yaml yet, but probably not worth a try.

helmuteke, 2024-05-21:
someone a other solution for this problem ?

geduxas, 2024-05-21:
> someone a other solution for this problem ?

Just downgrade esphome.. i have other products which have stopped working with latest esphome.. now i keep older version and don't upgrade

helmuteke, 2024-05-21:
> > someone a other solution for this problem ?
> 
> Just downgrade esphome.. i have other products which have stopped working with latest esphome.. now i keep older version and don't upgrade

But how can i downgrade please . . . it's HA addon

hn, 2024-05-21:
- Setup Build environment (Python for Windows, Debian in a VM/WSL, a Raspberry Pi or whatever you like)
- [Install ESPHome manually](https://esphome.io/guides/installing_esphome) but make sure you use `pip3 install esphome==2024.4.1` for the older version, double-check afterwards with `esphome version`
- `esphome compile solis-emw3080.yaml`
- If successful, the UF2 image can be found somewhere in path `./.esphome/build/ ... /.pioenvs/ ...`

(not tested, surely something is missing ...)

helmuteke, 2024-05-21:
i  found  this handy repo : just compiled my uf2

https://github.com/khenderick/esphome-legacy-addons?tab=readme-ov-file

timh77, 2024-05-27:
To fix this I removed the webportal access, it seems to bring down the firmware size enough to fix the issue.
So not a fix for those who reley on the webportal, but for those using ESP home/Home Assistant it works fine.

`# captive_portal:`

hn, 2024-05-27:
Thanks @timh77 :) I personally don't like these kinds of workarounds, but this is probably the best option we have today.

hn, 2024-06-06:
2024.5.5 seems to work (even with captive_portal=enabled) again. I haven't investigated this in detail to see if this is a coincidence or if there was a specific fix for it.

timh77, 2024-06-06:
I guess it's related to this change in 2024.5.4 (May 28)

[web_server_base] Bump ESPAsyncWebServer-esphome to 3.2.2 [esphome#6797](https://github.com/esphome/esphome/pull/6797) by [@jesserockz](https://github.com/jesserockz)

Previously it used version 3.2.0. Just a minor change was included, but I guess that resulted in a much smaller filesize.

commit 3c14dc7a (2024-06-07): Revert "With ESPHome 2024.5.0+ memory seems to be scarce"

Do not use ESPHome 2024.5.0 to 2024.5.4.

This reverts commit 19156c8282093f630dde9ab8bb10951f30e76976
and closes #32.
closed 2024-06-07 by hn