hn/ginlong-solis #34: Values Unknown : Inverter type definiton : 2030

helmuteke, 2024-05-29:
hello ,

after succesful flashing my Wifi stick , i receive for my second inverter unknown values.
The only value i see is Inverter type definition 2030

Is this type supported by the custom firmware or how to solve ?

hn, 2024-05-29:
A more general suggestion: when you report issues, you should provide the necessary information, i.e. in this case: what type of inverter is involved, which Yaml is used, debug logs etc.

helmuteke, 2024-05-29:
hello ,

My apologize.

The type of the inverter is :  S5-EH1P5K-L

i tried with the both YAML

This are the logs : 

[logs_solis-esphome-emw3080-inv_logs (3).txt](https://github.com/hn/ginlong-solis/files/15480258/logs_solis-esphome-emw3080-inv_logs.3.txt)

helmuteke, 2024-05-29:
reflashed , now it works

closed 2024-05-29 by helmuteke