hn/ginlong-solis #35: Unable to read inverter settings

nextguyover, 2024-06-21:
I'm currently using the ESPHome config provided here with a S5-GR1P5K inverter. All the 3000+ **read** registers are working correctly, all values are being read. However, I am not able to read from 3000+ **holding** registers. 

E.g. when attempting to read the time (holding reg 3000-3005, so 3000-1 = 2999 = 0xBB7), I get the following error:

```
[W][modbus_controller:030]: Modbus device=1 set offline
[D][modbus_controller:043]: Modbus command to device=1 register=0xBB7 countdown=0 no response received - removed from send queue
[W][modbus_controller:064]: Modbus device=1 back online
```

I'm wondering whether the register range for settings is possibly different for my model of inverter? I've also tried some of the other possible ranges for time from the documentation PDFs in the readme. I tried 33022-33027, which is the expected range for an ESINV, but this also returns the same error. Range 43000-43005 actually returns data, but the data here doesn't seem correct for a timestamp.

hn, 2024-06-21:
You're using the [inv.yaml](https://github.com/hn/ginlong-solis/blob/master/solis-modbus-inv.yaml) for the S5-GR1P5K, right?
INV type inverters do not have any registers in the ten-thousands range, to my knowledge.

The inv.yaml has some existing logic to read and set the time ([line 355](https://github.com/hn/ginlong-solis/blob/master/solis-modbus-inv.yaml#L355)) which works perfectly fine here. Try to de-adjust the inverter time via the LCD display, the yaml will correct it after some time.

nextguyover, 2024-06-21:
Yep, that's what I've been using for the past few months, however the inverter time never got set correctly during this period. 

I found out today that all holding registers in the 3000 range (including [time](https://github.com/hn/ginlong-solis/blob/master/solis-modbus-inv.yaml#L355), [limit output power](https://github.com/hn/ginlong-solis/blob/master/solis-modbus-inv.yaml#L27), and [working mode](https://github.com/hn/ginlong-solis/blob/master/solis-modbus-inv.yaml#L44) were not sending back data (which, for the time lambda means the inverter time never gets set either), and the above error was being printed.

Unsure if directly writing to the holding register(s) would work, I haven't tried this.

hn, 2024-06-26:
I'll leave this issue open for a while, please get back if you make any progress.

JGeeforce, 2024-10-07:
Question: I have a S5-GR3P 6K, will I have this issue too when I will flash ESP firmware? Or is Modbus not used in the default setting?