hn/ginlong-solis #38: Not getting data from RHI-3P10K-HVES-5G

MPW1412, 2024-06-23:
Hello,

I stumbled accross this neat project, because I want to integrate my RHI-3P10K-HVES-5G into my home assistant and get real time data instead of 5-6 min summaries via Alibaba cloud.

I flashed esphome on a d1_mini_pro  and integrated it with home assistant. That part works. I don't get any data from the inverter though.

The only response I got, is that it set the inverter type to 2060.

Am I on the right path here, what could I be missing?

Best,
Matthias

hn, 2024-06-24:
You are using the esinv.yaml, right? Have you checked the logs, normally the ModBus component displays useful information.

MPW1412, 2024-06-24:
Oh wow that works! I had completely overseen that option. Thank you!

closed 2024-06-24 by MPW1412