hn/ginlong-solis #41: No connection to ESPHome API possible

carstengl, 2024-06-28:
Hello,
I am using the solis adapter in Home Assistant with the ESPHome add-on and since the update of ESPHome from 2024.6.1 to 2024.6.4 the solis-esphome-esp8266.yaml has issues with the connection to the ESPHome API and does not work properly anymore. The change for the OTA settings I already implemented (adding the additional line with platform: esphome). Information from the log: (I replaced the IP of the adapter with "localIIP"):

"Can't connect to ESPHome API for solis-esphome-esp8266 @ localIP: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2 >, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='localIP', port=6053))]: [Errno 111] Connect call failed ('localIP', 6053) (SocketAPIError) "

Does anyone else experience the same issue with the 2024.6.4 release of ESPHome? Any idea that needs to be changed to get it running back? Only a switch back to ESPHome Version 2024.4.1 brought back the adapter, even with the 2024.6.1 ESPHome Version it was not working anymore.

Belaial, 2024-07-15:
Running Firmware: 2024.4.0 myself so can't say, all works perfect here (as you already noticed one Firmware: 2024.4.1).

I wont update this at all during the summer just to keeps things working, will look at upgrading closer to winter when there is not that much solar production to track in case something breaks.

hn, 2024-08-21:
Unable to reproduce - please reopen if you have any new information.

closed 2024-08-21 by hn

Belaial, 2024-10-29:
Today I accidentally pushed "Update" without thinking :)

Went from 2024.4.0 to 2024.10.2, 15 minutes running so far without issues.

I have also been checking output log from the stick and there is nothing out of the ordinary, everything just works as before, keep in mind only 15 minutes have passed but I doubt anything will break after 15 minutes.

When I rebooted the stick manually the following logs are presented

```
INFO Processing expected disconnect from ESPHome API for solis-emw3080 @ 192.168.0.67
WARNING Disconnected from API
INFO Successfully connected to solis-emw3080 @ 192.168.0.67 in 1.090s
INFO Successful handshake with solis-emw3080 @ 192.168.0.67 in 1.415s
```

So I assume it works just fine.