hn/ginlong-solis #46: Solis Data Logger is not showing data

atiffarrukh, 2024-08-25:
I am pretty sure its not the right place, but any help here would appreciated. Or if you can point me to right direction. Solis support is of no use.

Solis data logger is not getting any data from inverter.
[image]
Solis data logger version
[image]
Logs from Advanced -> Operation
```
*****************************************  
The collector.total_working_time    :38745 
The collector_record.startup_cnt    :78 
The collector_record.restart_cnt    :1 
The collector_record.restart_reason :27 
The ginlong_flag.inverter_update_fail:0 
The http_info.download_record	   :0 
***************************************** 
```
and
```
<-- stack heap free size (39672) -->

[2024-08-25 14:56:56] send MODBUS_35000_single
0A 04 88 B8 00 01 9B 34 

[2024-08-25 14:56:59] Collecte times expired .
modbus data collected fail ! 
send MODBUS_RS485_INQUIRE_OLD
7E 0A A1 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 AB 

[2024-08-25 14:57:02] Collecte times expired .
modbus data collected fail ! 
Begin to update data to the cloud .

<-- stack heap free size (39672) -->
```
response from `/inverter.cgi`
```
;0;0;0.0;0;0.000000;d;NO;
```

What i have noticed is that it does get data once every 5 minutes, but it does not keep displaying the data on web UI. It shows for a second or two then its gone.

atiffarrukh, 2024-08-25:
@hn , thanks, can you guide on how do i check that? I can't find anything related to mobdus on inverter.

KhurramFHassan, 2024-09-05:
I have the same problem. I have verified that the inverter modbus address is 1. I haven noticed in the logs (datalogger webpage Advanced-Operating link) that modbus communication is failing as shown in the attached picture. I know the datalogger is sending data to the SolisCloud app; so it is working. Is there any other way to enable modbus communication?

[image]

KhurramFHassan, 2024-09-05:
I contact local Solis support and the tech guy upgraded the datalogger software to version 00011204 (previously it was version 000111CF). This is made no difference as I still get zeroes in the inverter part of the "Status" page. In fact, it has made it worse because now I do not see anything in log (Advanced-Operating link). After I pointed it out to them, the support people have not contacted me again.

atiffarrukh, 2024-09-10:
@KhurramFHassan , how can is verify modbus address on solis inverter?

KhurramFHassan, 2024-09-20:
Sorry for the late reply. You can find the information in the datalogger manual. I am attaching a screenshot of the instructions here.

[image]

closed 2024-09-26 by hn