hn/ginlong-solis #48: Appreciate some help

geedsen, 2024-09-19:
I have 2 x Solis 10.0 kW Dual MPPT (3F) . Their dataloggers (LAN) dont work. Now I am planning to buy two S3-WIFI-ST dataloggers.
And hopefully they will let me integrate the inverters into HA. So I am going through the installations of the ESPHome and Solis Add-ons. Probably for  lots of HA people easy to do, but I got lost in the amount of documentation in specially for ESPHome. So I hope someone is willing to help me out here. 
I installed ESPHome and started it.
I added the two yaml files to the ESPHome directory (1 in subdir 'common')
Then I need to modify the configuration:

[image]

The api encryption key, I can just paste any Encryption key there? 
The wifi ssid and pwd, I assume is off the wifi near the inverters.
and the ap wifi ssid and pwd, where to they come from, do they come with the wifi dataloggers?
And is it straight in like  'key': 'value' ?

and I see this: 
[image]

Do I need to add these to the secrets.yaml as well? any ota_password will do?

And in the ESPHome documentation, I see this: 
[image]

but it does not mention what configuration file, is that the secrets.yaml?

Probably more questions later. Any help appreciated.

geedsen, 2024-09-19:
Found some information now how !secret refers a variable to the secrets file. That helps.

geedsen, 2024-09-19:
I added the api encryption key to the secrets,  restarted ESPHome, but when I ask for the key I get this:
[image]
Is it because it comes its 'a secret'?

geedsen, 2024-09-22:
Sorry for posting my comment under "Set the MCU to UART boot mode (pull TX pin low during boot)", I thought it would make sense to keep my expererience with the same issues at the same topic. 

So I am in the next phase of flashing the datalogger. Now I need to buy a usb to serial adapter. 
1) Will any one do, there are so many on amazon.nl?
2) Is this one probably a good (fast) one? DSD TECH SH-U09C5 USB naar TTL UART-converterkabel met FTDI-chipondersteuning 5V 3.3V 2.5V 1.8V TTL
3) Can I do this from Windows?
4) In the installation documentation it says this: Click the three-dots button, then "Install" and "Manual Download".
Wait for the compilation process to finish and download the "UF2 package"., does that mean the UF2 package is different for every installation and is generated specially for my configuration?

Thanks.

hn, 2024-09-22:
1./2. The serial adapter you pasted looks good. One never can say for sure if it will work. Some serial adapters do not supply enough power for the MCU -- but you can not "see" that in advance. Also put some male-female Dupont wires in your shopping cart.
3. I never did this but it should work.
4. Use two config/uf2 files with different `name` and `friendly_name`, otherwise ESPHome/HomeAssistant gets confused with two adapters.

**If you document everything you do** here with pictures, links, screenshots **so others can learn from it**, I'll help you if there are problems.

geedsen, 2024-09-24:
I also tried jumping GND - TXD while inserting the adapter, keo that connection for around 20 seconds. No luck :(

geedsen, 2024-09-24:
THe only thing I get is "Connecting to chip", Hardware reset, Software reset and this looping. How can I tell the 'pull' worked? I also found that using the Dupont pin on the other side of the stick might be too short to connect a female to it. It really is very lose. So if that does not connect the pull will fail as well.

geedsen, 2024-09-24:
Yep, that was the problem. I just kept two males against txd/gnd and pulled those. That worked.

geedsen, 2024-09-24:
[image]
Where exactly should this config be?

geedsen, 2024-09-24:
Hmm Flashing resulted in this: 
[image]
[image]
Is that OK?

hn, 2024-09-24:
Yes, just use two different yaml files. People tend to give them descriptive names like `solis-esphome-emw3080-garage.yaml`. For dutch people `solis-esphome-emw3080-campervan.yaml` might be an option.

Just check if flashing suceeded by powering the stick *without* TX pulled low. If everything went ok, you should see the ESPHome boot log in terminal. If not, try to flash again.

geedsen, 2024-09-24:
Hmm, maybe it did work: 
[image]

hn, 2024-09-24:
Looks good. Now you never should need to connect the serial adapter again. You can update via ESPHome OTA. Press the LOGS button to see the device logs.

Maybe the terminal uses the wrong baud rate, never used that.

geedsen, 2024-09-24:
The logs look good. Should I already see any values for the entities? It is still just connected to the adapter.
What I am really interested in is the entity that allows output limiting (like 80%) Should I see that entity in the logs?

geedsen, 2024-09-24:
Thanks for all your help BTW!

hn, 2024-09-24:
You'll not see any values unless the stick receives values via ModBus from a powered-on inverter (usually they power down at night). Go to HomeAssistant->Settings->Integrations->ESPHome to see the details then.

Consider to star this project if you like it.

geedsen, 2024-09-25:
The only part I am not seeing is this part:
[image]
I cannot find any of the entities in HA....

hn, 2024-09-25:
You need to add the device on the HomeAssistant->Integrations page, see my last post.

JGeeforce, 2024-10-06:
@geedsen Would it be possible to create a step by step manual? I really would like to use my sitck with ESPhome too.

So your USB serial from Amazon worked, did you by this one: [https://www.amazon.nl/DSD-TECH-SH-U09C5-UART-converterkabel-FTDI-chipondersteuning/dp/B07WX2DSVB/ref=sr_1_1?__mk_nl_NL=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=YQHJ3L560HRQ&dib=eyJ2IjoiMSJ9.tHkeOHA4qfOxRF3Z-K9veKBHEjeoGM8XHN6lg94uUICUJh-Go3svtzWxoqwHScMSHyX_dhBgAYNS4jNvrNFALCaU6747tdQtzqpRJRvrFG_sPaadDvUNV3-vfTy6dA8W9hKUjRAlnyDNHx3x_obN62lzLt9jkP7Y4BYJ_UVj6F3ipY20To3En3THVTynjakgnnp0qQVSRRJpDwJjKts9NtwcrJNIQ4v496WVeVk1dVxl_90sATnAu_VL2w33JS3kJ4vNzYulHBgmPlfgilhsxslS8KpwGCTI1qVxlJkBw4k.XGWLmWm0dmM9o1OrdOPTiLv1DI2xVViOl5gCz2ed7j8&dib_tag=se&keywords=DSD+TECH+SH-U09C5+USB+naar+TTL+UART-converterkabel+met+FTDI-chipondersteuning+5V+3.3V+2.5V+1.8V+TTL&qid=1728237481&sprefix=dsd+tech+sh-u09c5+usb+naar+ttl+uart-converterkabel+met+ftdi-chipondersteuning+5v+3.3v+2.5v+1.8v+ttl%2Caps%2C152&sr=8-1](url)

geedsen, 2024-10-06:
@JGeeforce Yes, that is the USB stick I used.
THe step by step guide @hn made is pretty accurate. So please follow that and anything unclear just ask.

JGeeforce, 2024-10-20:
> So I just made a copy of the yaml file and gave it a new name and friendly name. ![image](https://private-user-images.githubusercontent.com/1316595/370427628-cf25c666-875c-4ffb-b60c-ea28ce926230.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk0MzQ1NTcsIm5iZiI6MTcyOTQzNDI1NywicGF0aCI6Ii8xMzE2NTk1LzM3MDQyNzYyOC1jZjI1YzY2Ni04NzVjLTRmZmItYjYwYy1lYTI4Y2U5MjYyMzAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAyMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMjBUMTQyNDE3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YTY2NGU3NWFlNWNlY2IxYzdkOGE4NjE4MjE3NTQ1OTU5NmJhNGY0OTdlMGU4ODBkZmMwNTcyZmQ3NzY1NmY1OCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.2sXk-dYqT28YOnxCq5LDdz4SAutq6-c8uNPcI7FwjFU)
> 
> That's all, generate a new uf2 with these settings etc....?

Not sure about the different yaml files. Why is this and what should I do?

JGeeforce, 2024-10-28:
I connected everything and I am trying to read the flash to make a backup of the original firmware, but after several minutes I get a timeout, so reading the flash stops. Using the windows version. it stops around 30%. Tried several times....

115200
Realtek AmebaZ (or should that AmebaZ2 ?)

What goes wrong, would like to make a backup before flashing.

JGeeforce, 2024-10-28:
Baud rate tip!

As I had issues with backing up the original firmware I had to set the baud rate in the Windows version to auto! I had it set to 112500, but then I only got read time out. With auto it took 6 minutes and with success.

FlexxFR, 2025-03-07:
> Set your USB serial adapter to 3.3V (not 5V).
> 
> 1. Connect GND of adapter to GND of stick
> 2. Connect TX of adapter to RX of stick
> 3. Connect RX of adapter to TX of stick
> 4. Do not connect VCC yet
> 5. Connect **TX of stick** to GND ("pull TX low"). TX pin of stick is already occupied by the wire which you connected in step 3. I found it convinient to connect an additional wire to the Dupont pin on the other side of the TX hole.
> 6. Now connect VCC of adapter to VCC of stick, stick powers up ("pull TX low during boot")
> 7. After 2 seconds remove (only!) the TX-GND wire you connected in step 5 (stick is now in flash mode). Be careful not to wiggle the other cables, otherwise the stick will restart without entering flash mode.
> 8. Backup whole flash via ltchiptool
> 9. Flash UF2 image via ltchiptool
> 
> secrets.yaml: `wifi_ssid`, `wifi_password` = WiFi credentials for _your_ home network which you want the stick to connect to. `wifi_ap_ssid`, `wifi_ap_password`, = WiFi credentials for a 'rescue access point' to access the stick is not able to conect to your home network. https://esphome.io/components/wifi.html#access-point-mode `api_encryption_key` = 32-byte base64 encoded string, copy a random one from https://esphome.io/components/api.html `ota_password` = a random password for later updates. You never need to enter it manually, so choose a strong password.
> 
> All the above values are hardcoded into the UF2 file. Do not give the UF2 files to untrusted persons as they contain you WiFi password.
> 
> The screenshot you pasted ("LDO Mode BD_Info") is the boot log of the original vendor firmware.

Need some help with step 8.

Backup > is that the option "read flash" or "read rom"?
I also need to select a chip family (beken 72xx, Realtek AmebaZ, Realtek AmebaZ2)

I also only thing I get is "Connecting to chip", Hardware reset, Software reset and this looping. How can I tell the 'pull' worked?

After a normal boot I do see RTL8195A[driver]: set ssid [solis001]
and a scan of nearby ssid's. So serial connection and the tool seems to be working.

FlexxFR, 2025-03-07:
> Use "read flash" and "AmebaZ". You're done when you have a dump with exactly 8388608 bytes in size, see [here](https://github.com/hn/ginlong-solis?tab=readme-ov-file#replacing-the-main-application):
> 
> ```
> $ ltchiptool -V
> ltchiptool v4.10.1 # use at least this version
> $ ltchiptool flash read -d /dev/ttyUSB0 RTL8710B solis-s3-firmware-1012f.bin
> I: Connecting to 'Realtek AmebaZ' on /dev/ttyUSB0 @ 1500000
> I: Reading Flash (8 MiB)
> $ ls -l solis-s3-firmware-1012f.bin
> 8388608 # file size should be exactly this, otherwise something has gone wrong
> ```

Thnx, I;m trying to use the Wnidows GUI tool.
Still no luck.

It's stuck in "reading chip info", timer is not moving. no output on terminal. And after XX minutes
E: RuntimeError: No NAK for Loud-Handshake mode: b'\x00\x00\x00\x00\x00'

FlexxFR, 2025-03-07:
`D [   4915.345] (+3.516s) Started FlashThread
D [   4915.345] (+0.000s) Starting READ operation; file = D:\dump.bin, port = COM4 @ Auto
D [   4915.392] (+0.047s) Terminal closed
D [   4915.392] (+0.000s) Stopped TerminalThread
V [   4915.501] (+0.109s) OnUpdate(busy=set())
V [   4915.501] (+0.000s) OnUpdate(busy={'com4'})
D [   4915.517] (+0.016s) Connecting
S [   4915.517] (+0.000s) <- TX: b'\x18\x07\x18'
V [   4915.626] (+0.109s) OnUpdate(busy=set())
V [   4915.626] (+0.000s) OnUpdate(busy={'com4'})
V [   4915.626] (+0.000s) OnUpdate(busy={'com4'})
S [   4915.642] (+0.016s) -> RX: b'\x15'
S [   4915.689] (+0.047s) -> RX: b'\x15'
S [   4915.735] (+0.047s) -> RX: b'\x15'
S [   4915.782] (+0.047s) -> RX: b'\x15'
S [   4915.829] (+0.047s) -> RX: b'\x15'
S [   4915.876] (+0.047s) -> RX: b'\x15'
S [   4915.939] (+0.062s) -> RX: b'\x15'
S [   4915.985] (+0.047s) -> RX: b'\x15'
S [   4916.032] (+0.047s) -> RX: '@'
S [   4916.032] (+0.000s) -> RX: b'\x15'
S [   4916.079] (+0.047s) -> RX: b'\x15'
S [   4916.126] (+0.047s) -> RX: b'\x15'
S [   4916.173] (+0.047s) -> RX: b'\x15'
S [   4916.235] (+0.062s) -> RX: b'\x15'
S [   4916.235] (+0.000s) <- TX: '!'
S [   4916.235] (+0.000s) <- TX: b'\x05\x12'
S [   4916.282] (+0.047s) -> RX: b'\x06'
V [   4916.282] (+0.000s) -- UART: Port baudrate set to 460800
V [   4916.407] (+0.016s) OnUpdate(busy={'com4'})
V [   4916.407] (+0.000s) OnUpdate(busy={'com4'})
V [   4916.407] (+0.000s) OnUpdate(busy={'com4'})
S [   4916.392] (+0.109s) -> RX: '@'
S [   4916.392] (+0.000s) -> RX: b'\x15'
S [   4916.439] (+0.031s) -> RX: b'\x15'
S [   4916.485] (+0.047s) -> RX: b'\x15'
S [   4916.532] (+0.047s) -> RX: b'\x15'
S [   4916.579] (+0.047s) -> RX: b'\x15'
S [   4916.595] (+0.016s) <- TX: '!'
S [   4916.595] (+0.000s) <- TX: b'\x05\r'
S [   4916.626] (+0.031s) -> RX: b'\x06'
V [   4916.626] (+0.000s) -- UART: Port baudrate set to 115200
V [   4916.751] (+0.125s) OnUpdate(busy=set())
V [   4916.751] (+0.000s) OnUpdate(busy={'com4'})
V [   4916.751] (+0.000s) OnUpdate(busy={'com4'})
S [   4916.751] (+0.000s) -> RX: '@'
S [   4916.751] (+0.000s) -> RX: b'\x15'
S [   4916.798] (+0.047s) -> RX: b'\x15'
S [   4916.845] (+0.047s) -> RX: b'\x15'
S [   4916.892] (+0.047s) -> RX: b'\x15'
S [   4916.938] (+0.047s) -> RX: b'\x15'
S [   4917.001] (+0.062s) -> RX: '@'
S [   4917.001] (+0.000s) -> RX: b'\x15'
S [   4917.048] (+0.047s) -> RX: b'\x15'
S [   4917.095] (+0.047s) -> RX: b'\x15'
S [   4917.142] (+0.047s) -> RX: b'\x15'
S [   4917.188] (+0.047s) -> RX: b'\x15'
S [   4917.188] (+0.000s) <- TX: '!!'
S [   4917.188] (+0.000s) <- TX: b'\x07'
S [   4917.235] (+0.047s) -> RX: b'\x06'
D [   4917.235] (+0.000s) XMODEM: transmitting to 0x10002000
V [   4917.235] (+0.000s) push_timeout(1.0)
D [   4917.235] (+0.000s) Begin start sequence, packet_size=1024
D [   4917.235] (+0.000s) Begin start sequence, packet_size=1024
D [   4917.235] (+0.000s) standard checksum requested (NAK).
D [   4917.235] (+0.000s) standard checksum requested (NAK).
D [   4917.235] (+0.000s) send: block 1
D [   4917.235] (+0.000s) send: block 1
S [   4917.235] (+0.000s) <- TX: 02 01 FE 00 20 00 10 19  20 00 10 9B 21 00 10 EF  .... ... ...!...
S [   4917.235] (+0.000s) <- TX: 21 00 10 F5 20 00 10 EF  21 00 10 41 05 00 08 05  !... ...!..A....
[... 71 lines cut ...]
V [   4917.329] (+0.000s) pop_timeout()
V [   4917.329] (+0.000s) push_timeout(0.1)
S [   4917.345] (+0.016s) -> RX: b'\x18\rxmodem_log_open \n\r\rclose xModem Transfer ...\r\n\r'
S [   4917.735] (+0.391s) -> RX: 'AmbZ'
S [   4917.735] (+0.000s) -> RX: 54 6F 6F 6C 5F 4D 61 72  6B 65 72 21 FF 68 40 17  Tool_Marker!.h@.
S [   4917.735] (+0.000s) -> RX: 55 41 52 54 49 4D 47 5F  44 6F 77 6E 6C 6F 61 64  UARTIMG_Download
S [   4917.735] (+0.000s) -> RX: 20 32 0A 0D 0D 4F 70 65  6E 20 78 4D 6F 64 65 6D   2...Open xModem
S [   4917.735] (+0.000s) -> RX: 20 54 72 61 6E 73 66 65  72 20                     Transfer 
S [   4917.735] (+0.000s) -> RX: 6F 6E 20 4C 6F 67 20 55  41 52 54 2E 2E 2E 0D FE  on Log UART.....

after that and endless:

S [   4917.735] (+0.000s) -> RX: FF                                                .
S [   4917.782] (+0.047s) -> RX: FF                                                .
S [   4917.845] (+0.062s) -> RX: FF                                                .
S [   4917.892] (+0.047s) -> RX: FF                                                .
S [   4917.938] (+0.047s) -> RX: FF                                                .
S [   4917.985] (+0.047s) -> RX: FF                                                .
S [   4918.032] (+0.047s) -> RX: FF                                                .
S [   4918.079] (+0.047s) -> RX: FF                                                .
S [   4918.142] (+0.062s) -> RX: FF                                                .
S [   4918.188] (+0.047s) -> RX: FF                                                .
S [   4918.235] (+0.047s) -> RX: FF                                                .

`

hn, 2025-03-07:
The CEN pin of the MCU is not wired on the PCB, therefore we do not use it. Instead you disconnect/connect VCC to reset the MCU, which has the same effect.

If you see a continous stream of weird characters the MCU has sucessfully entered flash mode.

You can also try to lower the baud rate, maybe that helps.

According to LTchiptool authors, failing to flash is in most cases caused by a insufficient power supply (on your serial adapter). Try using another adapter and/or USB port. If flashing still fails, you might have to file an issue within the LTchiptool project, debugging those logs is out of scope here.

FlexxFR, 2025-03-09:
okay, so it seems i'm in the right direction.
Any more tips?

Do I need any specific USB serial settings?
BAUD rate, databits 8 partiy (none).

When using the GUI tool, during "reading chip info", I do see the green bar moving, giving the impression it does something, the elapsed timer is NOT running. Is this correct? (Debug logs shows only RX:FF for minutes), until after 10-15 minutes is restarts. I don't think its actually doing something

JGeeforce, 2025-03-09:
When using GUI in Windows, set baudrate to auto!  That solved the backing issue at my side.

FlexxFR, 2025-03-09:
> When using GUI in Windows, set baudrate to auto! That solved the backing issue at my side.

I did try that but I'm not sur ehow that works. When I checked the COM4 setting on Windows it was set to 9600.
I'm now experimenting with different speeds under Windows. 
[image]

But I still dont know if it works at all. The only confirmation I have is that I do see terminal output after a normal boot, but all other operations seems to "stuck" somewhere.

Did you saw the "elapsed" timer running?
I see this for minutes...and them I get an error.
[image]

FlexxFR, 2025-03-11:
Ok, I made some progress with an external powersuppy.
It will take some attempts but its downloading, however way too slow (25-30 minutes)

[image]

I purchased another adapter, the one mentioned above by @geedsen DSD TECH SH-U09C5 USB naar TTL UART.
This one works much more stable, I can easily start the read process, however it is still too slow. The stick "resets"after 15minutes but the download is not finished.

I tried many combinations of USB speeds settings in windows, itchiptool settings, auto, fixed, fastest time so far is an estimation of 25-30 minutes. Any ideas what can cause this?

Any settings to tune?

FlexxFR, 2025-03-11:
Succes!!

First I checked other (non USB3) ports on my desktop, same results. It works but slow. Tweaking different serial settings, advanced timings only makes things worse and even slower.

Then I used a different (much older laptop) and there it worked immediately :)
I checked the USB serial port settings but they were still on default settings (speed was still on 9600 bps) so I don't think any of those options were making any difference. 

Reading the 8MiB took about 7 minutes. No errors.

During the write I had the same errors described earlier (send error: expected ACKl got b'#x15 for block....)
But it just proceeded and was done within a minute, after the upload the view of the GUI tool switches to the terminal output and I immediately saw the ESPhome bootloader and connected to my wifi.

In home assistant it;s auto-discovered as ESPhome device.
OTA updates works from ESPhome.

Only thing to do is to reassemble the stick, install on the invertor and wait for the sun to test for actual values.

DriesA, 2025-03-25:
> I also tried jumping GND - TXD while inserting the adapter, keo that connection for around 20 seconds. No luck :(

Can I ask a more basic question: how do you remove the casing of the S3-WIFI-ST?

hn, 2025-03-25:
>Can I ask a more basic question: how do you remove the casing of the S3-WIFI-ST?

Search for "open the case" in the [README](https://github.com/hn/ginlong-solis/blob/master/README.md)

IErenas, 2025-03-29:
I am also having problems in step 8.

I bought this adapter https://www.amazon.es/dp/B07TXVRQ7V?ref=ppx_yo2ov_dt_b_fed_asin_title

Performing the following steps:

- Connect adapter to computer
- Connecting the cables as follows:
  - GND-GND
  - RX(Adapter)-TX(Stick)
  - TX(Adapter)-RX(Stick)
  - VCC-VCC
 
I would expect to see in the ltchiptool terminal the normal stick boot, but I don't see anything in the terminal.

If I try to perform step 8 “Set the MCU to UART boot mode”, when and perform a backup, I continually get the following log:

```
D: Connecting
D: Connecting
D: Connecting
D: Connecting
D: Connecting
D: Connecting
S: <- TX: b'\x18\x07\x18'.
S: <- TX: b'\x18x07x07x18'.
S: <- TX: b'\x18x07x07x18' S: <- TX: b'\x18x07x07x18'
S: <- TX: b'\x18x07x07x18'.
```

What I am doing wrong?

Many thanks!

IErenas, 2025-03-30:
Finally I managed to get it. I just change the adapter to 5V, just to mention that it is not recommended as @hn said multiple times.

FlexxFR, 2025-03-30:
> Finally I managed to get it. I just change the adapter to 5V, just to mention that it is not recommended as [@hn](https://github.com/hn) [said](https://github.com/hn/ginlong-solis/issues/48#issuecomment-2371866988) multiple times.

Keep this in mind, it might give problems when reading/writing later on.  I had to change the voltage to 5V also, but only with my cheap-ass adapter. With 3V it would not work, with 5V it did boot but serial signal was not stable. With an external 3V power supply it worked better and stable. 
But with the better module (you have the same it appears) 3V worked fine, no external power needed.

hn, 2025-03-31:
The EMW3080 MCU nativly uses 3v3 voltage -- and for me, the serial port connection has always worked with 3v3. That's why I recommend using 3v3, I just don't want anyone to blow up their board with higher voltages.

There may be a voltage divider on the board so using 5v is fine, I've never checked.

DriesA, 2025-04-02:
> Set your USB serial adapter to 3.3V (not 5V).
>
> 1. Connect GND of adapter to GND of stick
> 2. Connect TX of adapter to RX of stick
> 3. Connect RX of adapter to TX of stick
> 4. Do not connect VCC yet
> 5. [...]

Thanks for your help.
I tried following your guide, but I am stuck.

I have a FT232RL 3.3v-5v TTL USB Serial Port Adapter. I have set it to 3.3V (I checked with a multimeter).
I connected the cables in the order you described.

As soon as I connect the VCC, I hear a beep on my Windows PC (the one you hear when you connect new hardware) and a LED lights up on the S3-WIFI-ST. I guess that's a good sign. When I disconnect the TX-GND nothing happens (should the LED start flashing?).

In the ItchipTool (4.12), I am unable to connect to the S3-WIFI-ST. Even when I choose "Get Chip Info" it just remains "Connecting to Chip" or "Software reset" or "Hardware reset". I am also not sure which Chip Family I'm supposed to choose, but I tried all of them.

How can I further troubleshoot this?

hn, 2025-04-02:
**5v vs 3v3**:

If you follow the traces for the RX/TX pin (which are easily visible on the board), you will quickly realize that they are each connected in series with a resistor (R2, R4 with silk screen `30X` => 20 ohms) directly to the UART2_Log pins of the EMW. 20 ohms will likely not be enough to convert 5v to 3v3, so **I think it is advisable not to supply 5v level to the RX/TX pins**.

I was too lazy to trace the VCC pin within the circuit more closely. Since this is connected directly to the Exceedconn connector, which is known to supply 5v power, **5v will likely be fine there (for VCC)**.

JabeBRD, 2025-07-10:
Hello.
I was looking into flashing my S3-WIFI-ST dongle.
I allready have laying around two cables with FTDI chip. Are these suitable for this application?
First is old OBD2 to USB adapter, it has FT232BM chip. K-line and L-line (ISO9141) in use, would need +5V and +12V powersupply.
Second is TTL-232R-5V adapter cable. https://ftdichip.com/products/ttl-232r-5v/
TTL-232R-5V says `Max Power Output: +5V / 75mA` is this enought? This cable was used to flash motorcycle ECUs and 5V line was used to set ECU to flash mode.

hn, 2025-11-01:
> I think I'm clear about what registers I need, but regarding of the Modbus definition I need to understand first how to use correctly the variables: device_class, state_class, value_type, accuracy_decimals, filters or skip_updates.

@djramipro These settings are well documented in the (ESPHome) documentation. If you simply copy an existing block at first, it might even work by chance. And as long as you only read registers, you likely can't break anything.

GoldieLocks89, 2025-11-12:
> > > Baud Rate: 9600
> > > Parity: None
> > > Data bits: 8
> > > Stop bit: 1
> 
> I think I have the same configuration, but right now my inverter is turned off. I'll confirm tomorrow. In the meantime, I would recommend checking the wifi stick to make sure that the connections between the board and the connector and between the connector and the inverter are fine.

Great, thanks.

I've been flashing with it disconnected from the inverter, only connected over USB. Then disconnecting the USB and plugging into the inverter, the logs you are above are with it connected to the inverter only. I assume that isn't going to be my issue but let me know if you did differently and I'll try to copy your process!

MrJCub, 2026-11-10:
For anyone that has issues reading or writing, I found I had to do one extra step and it worked every time. I don't know if it's because my adapter is a cheap one from Ali Express, but after holding TX to ground when powering on the module, I started the read or write and it ACK, but sits there without going any further. At that point I briefly touch TX to ground again and then it continues and finishes successfully.