hn/ginlong-solis #52: Unable to find action with the name 'script.execute'. script.execute: modbus_activity

Apfelstruhdl, 2025-02-22:
INFO ESPHome 2025.2.0
INFO Reading configuration /config/esphome/esphome-modbus-solis.yaml...
Failed config

sensor.modbus_controller: [source /config/esphome/common/solis-modbus-esinv.yaml:264]
  platform: modbus_controller
  modbus_controller_id: modbus_master
  name: Total DC output power
  device_class: power
  state_class: measurement
  unit_of_measurement: W
  register_type: read
  address: 33057
  value_type: U_DWORD
  on_value: 
    then: 
      - 
        Unable to find action with the name 'script.execute'.
        script.execute: modbus_activity

hn, 2025-02-26:
`modbus_activity` is defined [here](https://github.com/hn/ginlong-solis/blob/master/solis-esphome-emw3080.yaml#L99) and you have probably mixed it up somehow.

Apfelstruhdl, 2025-02-26:
Indeed, I altered the ESPHome main yaml a bit and removed the modbus_activity part. Sorry for the confusion. It it still a bit unintuitive why the activity is triggered on a specific register read.

closed 2025-02-26 by Apfelstruhdl