hn/ginlong-solis #55: Question: How to turn Solis invertor ON/OFF

FlexxFR, 2025-04-27:
So, this is more a support question, not an issue with this project (as it all works fine).

Limit output power works fine but the inverter is a 0% not really off. Today I would rather shu the entire inv down for a couple off hours. 
In the Solis Modbus manual I did found address 3007 ON/OFF however with multiple values as option
0xBE-ON 
0xDE-OFF 
0x10—Night ON enable 
0x11—Night ON disable 
0x20—24H Consumption Enable 
0x21—24H Consumption Disable 

But I'm not sure how to implement this? Is this a value that is "writable".  I'm only interested in On and Off.
I'm probably not the first one to investigate this. Any suggestions?

hn, 2025-06-01:
@WesSec There is a "[Limit output power](https://github.com/hn/ginlong-solis/blob/master/solis-modbus-inv.yaml#L29)" register. If you set this to a very low value, you're effectively turning off the energy contribution that the inverter delivers to the grid.

WesSec, 2025-06-09:
@hn i found in another source that it is possible to turn off the inverter via modbus, although i think we dont want this, as it would be "destructive", when the inverter is off, it is probably also not powering the logging stick, resulting in not being able to turn it on again. But feel free to test and report back

https://github.com/incub77/solis2mqtt?tab=readme-ov-file#screenshots