hn/ginlong-solis #56: (Fix inside) S5-EH1P3.6K-L with ESINV: always reports Load, Grid and Battery faults, and incorrect Battery Energy during discharge

philpem, 2025-04-28:
Thanks for providing this - I've been using it with my new Solis S5-EH1P3.6K-L and it mostly works great, but I've found a few issues.

**The "Load Fault", "Grid Fault" and "Battery Fault" indicators are always showing "Problem"**

I checked the ESINV specification and it seems like these are inverted (e.g. Load Fault 1=the load is normal).
I fixed this by adding the following to these sensors:
```yaml
    filters:
      - invert:
```

**The 'Battery Power' sensor only reports positive values.**

This makes a battery discharge look like the battery is charging. It seems like this needs to be combined with "battery current direction" (if 'Not Charging' then Battery Power needs to be made negative) to produce accurate indications.

I fixed this by giving "Battery current direction" an ID and assigning a lambda to it:
```yaml
    lambda:
        if (id(battery_current_direction).state) {
            return x;
        } else {
            return -x;
        }
```

hn, 2025-04-28:
Thank you very much for your contribution, sounds reasonable. As I don't have an ESINV inverter myself, it would be perfect if you could submit a tested PR that includes these changes.

hn, 2025-04-29:
Feel free to submit anything that will improve this project. However, please keep all PRs clean, small and self-contained and I will be happy to accept them.

For the HA-related stuff I suggest to create a `contrib` subdirectory in this project, and then put files there with the naming scheme `ha-template-xy` or something like that. You could even add a readme `HA-TEMPLATES.md` if you want.

PR #57 (2025-04-29): Fix ESINV Battery/Grid/Load Fault and battery power reporting
Fixes #56 

  * ESINV Battery, Grid and Load Fault flags are inverse logic, ie 1=OK, but HA expects 1=problem. Add invert filters for these.
  * ESINV Battery Power is always positive regardless of power direction. Use the Battery Current Direction to make discharge show as negative power values.
closed 2025-04-29 by hn