hn/ginlong-solis #61: No Modbus Response from Solis 1P3.6K-4G Despite Correct Wiring

mwitlox, 2025-07-25:
Hi,
First of all, thank you for this great project.

I've built the ESP8266 module exactly as shown in the example setup. Initially, I received CRC errors, which I resolved by swapping the A and B wires. Now, the ESP8266 connects without CRC errors, but I consistently get the following in the logs:

[11:47:38][D][modbus_controller:039]: Modbus command to device=1 register=0xBBC no response received - removed from send queue
[11:47:40][D][modbus_controller:039]: Modbus command to device=1 register=0xBCD no response received - removed from send queue
[...repeats every cycle...]

My inverter is a Solis 1P3.6K-4G.
Unfortunately, there is no setting in the inverter menu to enable or configure Modbus – it appears to be always enabled or hidden from the user. I reached out to Solis support, and their only advice was to replace my current S3 WiFi stick with an S2 stick that has newer firmware supporting local Modbus. However, I would prefer not to go this route.

My goal is to completely bypass the SolisCloud platform, which is slow and unreliable, and instead collect data locally and control the inverter via Home Assistant using this ESP8266-based Modbus approach.

In addition to swapping A/B, I also reversed the TX and RX lines to rule out communication direction issues – unfortunately, no success.
I also tried multiple values for the Modbus address in the config:

modbus_controller:
  - id: modbus_master
    address: 1

I’ve tested address values such as 1, 2, 3, and 247, but still receive no response.

Has anyone successfully connected to this specific inverter model (1P3.6K-4G) using this method?
Do I need to change any register, address, or configuration to get a response from the inverter?

Any insights or suggestions are highly appreciated.

.yaml:

esphome:
  name: esp-solis-inverter
  friendly_name: Solis Inverter

esp8266:
  board: d1_mini

logger:

packages:
  # Uncomment exactly one line depending on your device type
  inv:   !include common/solis-modbus-inv.yaml
  #esinv: !include common/solis-modbus-esinv.yaml
  #epm:   !include common/solis-modbus-epm.yaml

  # Uncomment for optional INV inverter addon  
  #inv-acr10r-1ph: !include common/solis-modbus-inv-addon-acr10r-1ph.yaml

time:
  - platform: homeassistant
    id: network_time

api:
  encryption:
    key: !secret api_encryption_device3

ota:
  platform: esphome
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "esp-solis-inverter"
    password: !secret fallback_ap_password

captive_portal:

script:
  - mode: restart
    id: modbus_activity
    then:
      # e.g. blink led

uart:
  id: modbus_uart
  tx_pin: D6
  rx_pin: D5
  baud_rate: 9600

modbus:
  #flow_control_pin: D7
  send_wait_time: 300ms
  id: modbus_hw

modbus_controller:
  - id: modbus_master
    address: 1
    modbus_id: modbus_hw
    setup_priority: -10

text_sensor:
  - platform: wifi_info
    ip_address:
      name: "IP Address"
    ssid:
      name: "Connected SSID"
    mac_address:
      name: "MAC Address"
    dns_address:
      name: "DNS Address"

button:
  - platform: restart
    name: "Restart device"

hn, 2025-07-27:
>replace my current S3 WiFi stick

Do you own an S3 stick, does it work?

>Any insights or suggestions are highly appreciated.

The RS485 adapter pcb usually has an RX/TX led, do they both light up when the ESP sends a request?

The solis inverter (via LCD menu) should have a way to set the inverter's ModBus address. Double check that it is set to "1".

mwitlox, 2025-07-27:
> Do you own an S3 stick, does it work?
>
>
> The RS485 adapter pcb usually has an RX/TX led, do they both light up when the ESP sends a request?
>
> The solis inverter (via LCD menu) should have a way to [...]

I have a working S3 stick :)

Only the TX flashes when it is reading data.

I just found something in the settings menu. The option is called Set Address. There is nothing in the menu that says Modbus, but I guess I'm on the right track. The address is set to 01.

mwitlox, 2025-07-27:
Solved. I used the following pin layout instead of yours.

https://github.com/hvoerman/solis2mqtt

It seems that they are not all the same. The data is now coming in nicely!

Now I'm going to continue with the on/off and the limiter. Not entirely clear what the possibilities are. Switching off was possible via the unreliable Solis Cloud. I'll try the limiter when there is enough sun. All for when net metering stops in the Netherlands.

Unfortunately, my house already had a Solis String Inverter system. I added an Enphase system. This combination is not ideal, but with Home Assistant, I should be able to create something nice and regulate self-consumption as much as possible.

closed 2025-07-27 by mwitlox